asalamon74/pktriggercord #6: K50 Support Question

farbodab, 2016-02-19:
Hello,

I have a k50 and this is the only software I've found that seems very promising. Let me know what I can do to help you with improving support for the K50. I currently cannot take pictures in bulb mode. I can in manual mode but I cannot change the shutter speed or the bulb exposure mode.

asalamon74, 2016-02-21:
What operating system do you use?

First, please set the camera to bulb mode and execute the following command and copy the result here:

pktriggercord-cli --debug --status_hex

Please also try the following:

pktriggercord-cli --debug -t 32 -o test.dng

farbodab, 2016-02-21:
I've used both linux and windows with the same results. Here are the results of commands in linux terminal:

jharvard@appliance (~/Desktop/pktriggercord): sudo ./pktriggercord-cli --debug --status_hex

	Debug messaging is now enabled.
./pktriggercord-cli 0.83.01 
model (null)
device (null)
[C]	plsr_init()
driveNum:2
	Checking drive:  sg0 VMware, 
 VMware Virtual S

	Checking drive:  sg1 PENTAX  
 DSC_K-50        

	Found camera PENTAX  
 DSC_K-50        

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [1C 00 00 00  00 00 01 00]
[R]				 => [1C 00 00 00]
[C]			read_result(0x3, size=28)
[S]					 >>> [F0 49 00 00  1C 00 00 00]
[S]					 <<< [00 00 00 02  00 00 00 00  02 00 02 00  80 80 00 01 
					      00 00 00 00  00 00 00 03  00 00 00 1B]
[R]				 => [00 00 00 02  00 00 00 00  02 00 02 00  80 80 00 01
				     00 00 00 00  00 00 00 03  00 00 00 1B]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [1C 00 00 00  00 00 01 00]
[R]				 => [1C 00 00 00]
[C]			read_result(0x3, size=28)
[S]					 >>> [F0 49 00 00  1C 00 00 00]
[S]					 <<< [00 00 00 02  00 00 00 00  01 00 02 10  80 80 00 01 
					      00 00 00 00  00 00 00 00  00 00 00 1B]
[R]				 => [00 00 00 02  00 00 00 00  01 00 02 10  80 80 00 01
				     00 00 00 00  00 00 00 00  00 00 00 1B]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [08 00 00 00  00 00 01 00]
[R]				 => [08 00 00 00]
[C]			read_result(0x3, size=8)
[S]					 >>> [F0 49 00 00  08 00 00 00]
[S]					 <<< [00 01 2F B6  00 00 02 21]
[R]				 => [00 01 2F B6  00 00 02 21]
[... 732 lines cut ...]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 81]
[R]				 => ERROR: 0x81

asalamon74, 2016-02-24:
The Bulb mode setting on the camera is correctly recognized, but the camera does not accept the bulb start command. It might be firmware limitation, or the K-50 requires a different way to start bulb exposure.

I'll add this info to the BUGS file and keep the issue open, maybe later someone with a K-50 camera will send a patch.

Blueshawk, 2017-03-11:
:picks up the gauntlet:

asalamon74, 2017-03-12:
@Blueshawk Thanks, waiting for the pull request. ;)

asalamon74, 2017-07-17:
I've fixed a similar bug for K-70, hopefully the solution will work also for K-50.

I need the output of the following command twice. Once when the camera is set to BULB but the timer is not set, second when the camera is set to BULB and the timer is set to a long exposure (preferably 5 minutes). Please also tell me the exact time it was set:

`pktriggercord-cli --settings_hex --debug > debug.log 2>&1`

Blueshawk, 2017-07-17:
Woot! I'm on it! Stopping other work. I got stuck on this after helping out at libgphoto with an apparent command offset issue. Fingers crossed! I have a couple of procedural questions , to make sure we are on the same page for testing.
1. Which menu setting should I set the bulb type to in the menu? press and hold or press press? Or will it not matter? 
2.  Which dial setting should I set it to? "B" bulb seems like a fair choice but I wanted to be sure since It can also be set to bulb in user modes using firmware.

Blueshawk, 2017-07-17:
oh argh!...I checked the manual and the k70 has an internal timer for bulb mode. The k50 doesn't have this. I does however have the two modes I mentioned above. I'll go ahead and get you hex from each one.
Here's the one I got from the first part:
[bulb1debug.log.txt](https://github.com/asalamon74/pktriggercord/files/1154171/bulb1debug.log.txt)
And the second one:
[bulb2debug.log.txt](https://github.com/asalamon74/pktriggercord/files/1154172/bulb2debug.log.txt)

Hope this helps. :)

asalamon74, 2017-07-18:
@Blueshawk 

Setting the dial to "B" is the best, I'm not sure about the user modes.

K-70 has two bulb modes (green button can change between them) one is the indefinite time (press-press or press-hold to specifies the length) and the other is the internal timer mode where you specify in advance the exposure time.  In this mode a single press is enough to take a bulb picture.

A K-1 tester found out that bulb works correctly in the internal timer mode. So my solution was to change the camera to the interval timer mode from the code, specify the length from the code and send a single press command. (These changes are camera-specific that's why I was asking for more debug info).

If K-50 has no internal timer mode then this patch cannot be applied here. :(

I'll check the press-press and press-hold modes of my K-70 camera and maybe come up with a different solution. It's a long shot.

asalamon74, 2017-07-22:
@Blueshawk Thanks to your 2 debug logs, I've commited a change, now the program is able to read to bulb mode of K-50 (press-press or press-hold). Please try it. It does not really help to fix the bulb problem, but it might be useful later.

Blueshawk, 2017-07-22:
I went ahead and made new logs with the settingshex and with -t60  
 In mode 1 the shutter opens but closes right away ignoring any delay, and does not do the second one due to the error(presumably) as before.
In mode 2 it is a different story, The shutter locks open and stays, never finishing or writing a file

[rambling a bit jic something helps] Seems like I got it to do the shutter twice in this mode in my own hacking by ignoring the status and just retriggering shutter after the delay and creating two files of short shutter time. which shows the camera is happy at that point in the process, just didn't grok the command. Bulb vs. shutter.  I noticed that bulb() includes shutter in this arrangement which might explain that. Maybe in early models the shutter command was be enough to trigger the bulb routine? or maybe the other way around. I wonder if in mode two there is a shutter open, and then a second shutter call would close it?  Yes it is confusing and i'm sure it's more complicated than that.  I would do more hacking on my own on this but i'm busy trying to get experiments ready to try to do crazy eclipse photography...because I'm apparently slightly insane. :) What we really need is the guys who originally cracked the protocol to reget the command structure but apparently that has gotten harder and more encrypted.

[debugbulb1.log.txt](https://github.com/asalamon74/pktriggercord/files/1167440/debugbulb1.log.txt)
[debugbulb1nohexsettingst60.log.txt](https://github.com/asalamon74/pktriggercord/files/1167441/debugbulb1nohexsettingst60.log.txt)
[debugbulb2.log.txt](https://github.com/asalamon74/pktriggercord/files/1167442/debugbulb2.log.txt)
[debugbulb2nohexsettingst60.log.txt](https://github.com/asalamon74/pktriggercord/files/1167443/debugbulb2nohexsettingst60.log.txt)

asalamon74, 2017-07-26:
I've tried press-press mode and sent a shutter command (pslr_shutter) waited for a few seconds and sent a second pslr_shutter command but it was not working. The second call of pslr_shutter gave me error.

Blueshawk, 2017-07-26:
Yeah there was something in the code that tripped it, I wish I could remember how I tricked it into doing that. I'll look it over again and see if I can duplicate it. Might be nothing but it's a squirrel to chase anyway. :D