asalamon74/pktriggercord #8: Issues with my K-3

crispinus2, 2016-05-23:
pktriggercord seems to be completely incompatible to my K-3 (firmware version is 1.21). pktriggercord recognizes the camera correctly and I can see the status information, but I'm not able to set any parameters (like exposure mode, shutter speed, iso or aperture value) and the camera shows impossible values for shutter speed and aperture mode if set to Manual exposure by the mode wheel. The debug output of pktriggercord (attached)
[debug.txt](https://github.com/asalamon74/pktriggercord/files/276979/debug.txt)

 clearly shows something going wrong, but I have no clue what goes wrong or how to fix it. Help is appreciated.
[debug.txt](https://github.com/asalamon74/pktriggercord/files/276980/debug.txt)

asalamon74, 2016-05-26:
You're not the first one reporting similar K-3 problems. It seems to me the program works for some K-3 users and does not work for others. 

You can check the [issue](https://github.com/asalamon74/pktriggercord/issues/2) opened by the original K-3 developer. Some of the features were not working for him, but he was able to take (sometime not perfect) images.

It might me a firmware-specific problem.

I don't have a K-3 for testing, maybe some K-3 user will send a patch someday.

56Tbird, 2021-03-31:
Has a patch been sent for the K-3. I am not even getting to first base with the program, following all the instructions gets me a blank screen and a non functioning camera. help?

asalamon74, 2021-04-05:
@56Tbird Small fixes was applied but K-3 support is still not the best. Here you can find a few tricks that might help: http://pktriggercord.melda.info/tips_and_tricks.html 

What operating system do you use? What is the debug output of the program?

billh8, 2022-07-25:
Any progress for full K-3(original) support?

I see comments on "bulb mode" & "shuttle speed" problems. But I can't take pictures at all!

Running GUI mode on Windows 7 shows "JSON: Cannot find camera model" message. Status output is fine but no "settings info." Hitting the Focus button causes the camera to focus but hitting the take picture button will hang the program.

I have attached Windows 7 command line outputs for the following 2 commands:
pktriggercord-cli --debug --status_hex
pktriggercord-cli --debug --file_format JPEG -o test.jpg

I hope this sheds more light on the problem.

Thanks!
[status_hex_debug.txt](https://github.com/asalamon74/pktriggercord/files/9177247/status_hex_debug.txt)
[cli_take_picture_debug.txt](https://github.com/asalamon74/pktriggercord/files/9177248/cli_take_picture_debug.txt)

billh8, 2022-07-26:
Following up on yesterday's comments.

The problem I'm encountering doesn't seem to be related to firmware version. I started with v1.43 & successively downgraded it to v1.20 & the behavior at each firmware revision seems to be the same.

HTH!

asalamon74, 2022-08-07:
@billh8 This is a useful information, at least it shows that the problem is not firmware-related. Unfortunately it does not help to find the problem.

The "JSON: Cannot find camera model" is not too important. Most of the information can be read from the status buffer, like this one from your log:

```
current iso                     : 1600
current shutter speed           : 1/15
camera max shutter speed        : 1/8000
```
but some information can be acquired by reading settings. The raw data can be read using `--settings_hex`. The decoding rules are defined in the `pentax_settings.json` file, but only a few models are supported. 

I know that it's not exactly the same model, but please check #35 , for K-3 II lens correction had to be turned off.

billh8, 2022-08-10:
Thanks for the lead. After disabling all the lens correction options, I was able to snap a shot & have it download via the command line interface.

However, I ran into inconsistent behavior when I tried to set ISO, aperture & shutter speed via the command line. At first, the only problem I encountered was not being able to set ISO; aperture & shutter speed responded correctly.

The K-3 was running firmware v1.21 for the above tests. I updated the firmware to the latest v1.43. Now via command line I was only able to control the aperture. I couldn't set the ISO and the shutter speed was stuck at 30"!

I downgraded the firmware back to v1.21. Unfortunately, it didn't behave as seen previously. Now the shutter speed is always 30", same as with the latest firmware.

At least I can get snapshots via the command line. I just have to preset ISO, aperture & shutter speed prior to tethering.

billh8, 2022-08-10:
Sorry, should have searched for shutter speed issues before posting my last comment. Saw in issue [#51](https://github.com/asalamon74/pktriggercord/issues/51) that connecting the K-3 after switching on resolved the shutter speed problem.

Did that for both firmware v1.21 & v1.43. I was able to control both aperture & shutter speed now.

I also saw that ISO can't be changed while connected, but the camera would be set to the last ISO value requested before disconnecting.