asalamon74/pktriggercord #9: Increase output to file speed

nanowish, 2016-05-27:
Hello,

currently output to file (/get from buffer) take between 4 and 7 seconds (with my K-5) and seem too long than write to SDcard when I take picture directly from APN. Any tips for increase write speed ? tested write in tmpfs, ssd, sdcard, connecting to USB2.0, USB3.0 ... the bottleneck seem to be from APN buffer to computer. When triggercord write the file can't do more.

tested with:
`pktriggercord-cli --model=K-5 -i 200 -t 1/2 -r 10 --flash_mode=Manual-RedEye --aperture=4.0 --reconnect -o fwritetest -f`

`pktriggercord-cli --model=K-5 -i 200 -t 1/2 --flash_mode=Manual-RedEye --aperture=4.0 --reconnect -o fwritetest -f`

`pktriggercord-cli --model=K-5 -i 200 -t 1/120 --flash_mode=Manual-RedEye --aperture=4.0 --reconnect -o fwritetest -f`

`pktriggercord-cli --model=K-5 -i 200 -t 1/160 --aperture=4.0 --reconnect -o fwritetest -f`

`pktriggercord-cli --model=K-5 -i 200 -t 1/160 --aperture=4.0 -o fwritetest -f`

Thank for your work and this software :+1:

asalamon74, 2016-05-29:
4-7 seconds for a jpeg? How long does it take to download a dng?

nanowish, 2016-06-15:
Yes for a jpeg.

DNG take more time:
```
time pktriggercord-cli -r 12 --model=K-5 -i 200 -t 1/160 --aperture=4.0 -o fwritetestdng.dng -f --file_format=DNG
pktriggercord-cli: K-5 Connected...

real    0m11.010s
user    0m0.010s
sys     0m0.360s

file fwritetestdng.dng-0000.dng
fwritetestdng.dng-0000.dng: TIFF image data, big-endian, direntries=38, height=0, bps=0, compression=none, PhotometricIntepretation=RGB, manufacturer=PENTAX             , model=PENTAX K-5         , orientation=upper-left, width=0

-rw-r--r--   1 root root  12M Jun 15 17:22 fwritetestdng.dng-0000.dng
```

JPEG test:
```
time pktriggercord-cli -r 12 --model=K-5 -i 200 -t 1/160 --aperture=4.0 -o fwritetestjpeg.jpeg -f --file_format=JPEG
pktriggercord-cli: K-5 Connected...

real    0m7.920s
user    0m0.000s
sys     0m0.210s

-rw-r--r--   1 root root 6.9M Jun 15 17:23 fwritetestjpeg.jpeg-0000.jpg

root@orangepipc:~# file fwritetestjpeg.jpeg-0000.jpg
fwritetestjpeg.jpeg-0000.jpg: JPEG image data, Exif standard: [TIFF image data, big-endian, direntries=17, manufacturer=PENTAX             , model=PENTAX K-5         , orientation=upper-left, xresolution=258, yresolution=266, resolutionunit=2, software=K-5 Ver 1.16           , datetime=2016:06:15 17:22:45], baseline, precision 8, 4928x3264, frames 3
```

nanowish, 2016-06-17:
I added some informations.

Select buffer return many ERROR and takes 3-4 secondes:

```
Test:
root@belenos:/mnt/pictures# /usr/bin/pktriggercord-cli --model=K-5 -i 400 -t 1/80 -r 16 --flash_mode=Manual-RedEye --aperture=4.0 --reconnect -o testphoto -f --debug |& gawk '{ print strftime(), $0 }' | egrep -E "Select buffer 0,5,0,0|0: Addr: 0x0 Len:"

Fri Jun 17 23:13:57 CEST 2016 		Select buffer 0,5,0,0
Fri Jun 17 23:14:01 CEST 2016 	0: Addr: 0x0 Len: 0(0x00000000) B=1

Other test:
Fri Jun 17 23:14:14 CEST 2016 		Select buffer 0,5,0,0
Fri Jun 17 23:14:17 CEST 2016 	0: Addr: 0x0 Len: 0(0x00000000) B=1

Other test:
Fri Jun 17 23:14:25 CEST 2016 		Select buffer 0,5,0,0
Fri Jun 17 23:14:29 CEST 2016 	0: Addr: 0x0 Len: 0(0x00000000) B=1

```

Debug output of "[Select buffer](https://github.com/asalamon74/pktriggercord/blob/08d4f22286501c1fbaf43ddf48f70af45b1cdcda/pslr.c#L1208-L1224)"

```
---------------------------
	p->status.bufmask = 1
		Select buffer 0,5,0,0
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x0, 0x5, 0x0, 0x0})
[S]					 >>> [F0 4F 00 00  10 00 00 00]
[S]					 >>> [00 00 00 00  00 00 00 05  00 00 00 00  00 00 00 00]
[C]			command(fd=3, 2, 1, 10)
[S]					 >>> [F0 24 02 01  10 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[C]		ipslr_buffer_segment_info()
[C]			command(fd=3, 4, 0, 0)
[S]					 >>> [F0 24 04 00  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [10 00 00 00  00 00 01 00]
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< [00 00 00 02  00 00 00 01  00 00 00 00  00 00 00 00]
[R]				 => [00 00 00 02  00 00 00 01  00 00 00 00  00 00 00 00]
	0: Addr: 0x0 Len: 0(0x00000000) B=1
```

asalamon74, 2016-06-30:
My tests also shoes the select buffer takes quite a long time. The errors are not "real" errors. It only means that we have asked the camera to switch to the buffer, but the camera is not yet ready. If I don't wait till the switch is successful  I cannot download the image later.

I've modified the poll interval so I check the result of the buffer switch more often. It increased the speed a bit (maybe 5%) I'm afraid I cannot increase the speed more.

closed 2016-06-30 by asalamon74