asalamon74/pktriggercord #10: Support for K-1?

horemachet, 2016-06-24:
Hi Folks,

are there any plans to support the new Pentax K-1 already?
Do you neet some testing support?

Regards,

Karsten

asalamon74, 2016-06-26:
Someone was planning to add support but haven't sent me the patch yet.

If you're ready to test it I can try to add support. What operating system do you use?

horemachet, 2016-06-26:
Hi,

I would be glad to help. I am using openSuSE Leap 42.1.

Best Regards, Karsten

asalamon74, 2016-06-29:
I've commited limited support for K-1, please clone the latest code and compile it.

The program should recognize the camera. First I suggest to test the following command, please copy the output here:

pktriggercord-cli --status_hex --debug

Knubbel, 2016-06-29:
Hi, here is the output using the windows version I found here.

Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.00
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=2c, 0, 1, 0)
[C]                     get_result(0x2c)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x2c, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 0
0 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=2c, 0, 0, 4)
[C]                     get_status(0x2c)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=2c, 0, 1, 0)
[C]                     get_result(0x2c)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x2c, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 0
0 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=2c, 0, 4, 0)
[C]                     get_result(0x2c)
[R]                              => [08 00 00 00]
[C]                     read_result(0x2c, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092

Unknown Pentax camera.

asalamon74, 2016-06-29:
The latest release (0.84.00) does not support K-1, only the development (0.84.01) version. Since it's quite difficult to compile the program for windows I attach a windows test version here.

[pktriggercord-0.84.01-win.zip](https://github.com/asalamon74/pktriggercord/files/339664/pktriggercord-0.84.01-win.zip)

Knubbel, 2016-06-29:
Thanks. This is the output:

```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 0
[C]                     read_result(0x30, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (420 bytes more)]
        init bufmask=0x0
[C]             ipslr_cmd_00_09(0x2)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x2})
[C]                     command(fd=30, 0, 9, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[... 116 lines cut ...]
light meter flags               : 0
ec                              : unknown
custom ev steps                 : 1/2
custom sensitivity steps        : 0
exposure mode                   : 0 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : MF
af point select                 : Auto-5
selected af point               : 0
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : unknown
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : M7B7
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : M-42 or No Lens
battery                         : 0.00V 0.00V 0.00V 0.00V

```

horemachet, 2016-06-29:
I am on a business trip until Friday and don't have the camera with me. 
I just set up the test einvironment though. 
I'll give it a trial on Friday and post the results here.

Best regards,

Karsten

asalamon74, 2016-07-02:
I've modified the code, I attach the new windows test version.
[pktriggercord-0.84.01-win.zip](https://github.com/asalamon74/pktriggercord/files/344431/pktriggercord-0.84.01-win.zip)

Please execute the same command and also give me the following information:

- The name of the lens

- The focal length used

horemachet, 2016-07-02:
Hi Folks,

I just gave it a trial. It seems it doesn't recognizes the camera.
Camera is set to PTP-Mode.
I did a trial with MSC Mode but with no effect.

`
./pktriggercord-cli --status_hex --debug
Debug messaging is now enabled.
./pktriggercord-cli 0.84.01 
model (null)
device (null)
[C]     plsr_init()
driveNum:2
        Checking drive:  sg0 ATA     
 WDC WD5000LPVT-2

        Checking drive:  sg1 Optiarc 
 DVD RW AD-7700H 

        camera not found
diff: 0.000357
sleep 1 sec
[C]     plsr_init()
driveNum:2
        Checking drive:  sg0 ATA     
 WDC WD5000LPVT-2

        Checking drive:  sg1 Optiarc 
 DVD RW AD-7700H 

        camera not found
diff: 1.000800
sleep 1 sec
[C]     plsr_init()
driveNum:2
        Checking drive:  sg0 ATA     
 WDC WD5000LPVT-2

        Checking drive:  sg1 Optiarc 
 DVD RW AD-7700H 

        camera not found
diff: 2.001229
sleep 1 sec
^C

`

asalamon74, 2016-07-02:
The camera should be set to MSC mode. Can you test it as root?

Knubbel, 2016-07-02:
That is the output using the new windows version you posted above:

```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  50 00 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  a0 0f 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  01 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 03 00 00 00  06 00 00 00 04 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 05 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 02 00 00 00 04 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  00 00 00 00 07 00 00 00
0x00e0 | 00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x00f0 | 01 00 00 00 01 00 00 00  00 00 00 00 88 13 00 00
0x0100 | 08 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0110 | 01 00 00 00 50 00 00 00  1c 00 00 00 0a 00 00 00
0x0120 | 00 00 00 00 0a 00 00 00  00 00 00 00 0a 00 00 00
[... 175 lines cut ...]
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 4 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.C
af point select                 : Single
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : M-42 or No Lens
battery                         : 5.60V 6.49V 0.00V 0.00V

```

asalamon74, 2016-07-02:
@Knubbel Getting better. I attach a new version.

[pktriggercord-0.84.01-win.zip](https://github.com/asalamon74/pktriggercord/files/344458/pktriggercord-0.84.01-win.zip)

A few important fields still need to be found. Please set the camera to M mode and shutter speed to 1/1000 and execute the same command, after that to the same for 1/2000. The difference should show the place of the shutter speed field.

PcS: Similar thing has been reported for several cameras with a top LCD. My K-x has no such thing so I cannot debug it.

Knubbel, 2016-07-02:
That is the output at 1/1000 (setting the camera must be done prior to pluging in the USB cable as the camera is inaccesible afterwards in PC-Sync mode).

pktriggercord-cli.exe --status_hex --debug
```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  e8 03 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 19 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  01 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 03 00 00 00  06 00 00 00 04 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 08 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 01 00 00 00 00 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  00 00 00 00 07 00 00 00
0x00e0 | 00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x00f0 | 01 00 00 00 01 00 00 00  00 00 00 00 88 13 00 00
0x0100 | 08 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0110 | 01 00 00 00 e8 03 00 00  1c 00 00 00 0a 00 00 00
[... 169 lines cut ...]
raw format                      : DNG
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 6 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Auto-5
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 5.60V 6.46V 0.00V 0.00V
```

Knubbel, 2016-07-02:
And now again in M mode but with shutter at 1/2000:
```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  d0 07 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 19 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  01 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 03 00 00 00  06 00 00 00 04 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 08 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 01 00 00 00 00 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  00 00 00 00 07 00 00 00
0x00e0 | 00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x00f0 | 01 00 00 00 01 00 00 00  00 00 00 00 88 13 00 00
0x0100 | 08 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0110 | 01 00 00 00 d0 07 00 00  1c 00 00 00 0a 00 00 00
0x0120 | 00 00 00 00 0a 00 00 00  00 00 00 00 0a 00 00 00
0x0130 | 01 00 00 00 40 1f 00 00  00 19 00 00 00 00 00 00
0x0140 | 80 00 00 00 00 00 00 00  1c 00 00 00 0a 00 00 00
[... 162 lines cut ...]
raw format                      : DNG
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 6 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Auto-5
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 5.60V 6.46V 0.00V 0.00V
```

Knubbel, 2016-07-02:
Just as a side comment:
I chose the fixed center AF point which pktriggercord reports as 16384.
maybe you take a look at how Phil Harvey from exiftool interprets things. He calls it "Fixed Center;0" for the K-1. The numerical values for AF point selected in the exifdata for this shot are "65534;0".

asalamon74, 2016-07-02:
@Knubbel I've fixed a few more things, hopefully most of the important fields will be parser correctly in this version. Plase test it.

[pktriggercord-0.84.01-win.zip](https://github.com/asalamon74/pktriggercord/files/344505/pktriggercord-0.84.01-win.zip)

I cannot find the value 65534 in the status buffer. 

You might test taking a picture using the following command:

./pktriggercord-cli --debug --file_format dng -o test1.dng

horemachet, 2016-07-02:
Okay, did it as root and in MSC-Mode.
Result is definitely different :-)

`
./pktriggercord-cli --status_hex --debug
Debug messaging is now enabled.
./pktriggercord-cli 0.84.01 
model (null)
device (null)
[C]     plsr_init()
driveNum:4
        Checking drive:  sg0 ATA     
 WDC WD5000LPVT-2

        Checking drive:  sg1 Optiarc 
 DVD RW AD-7700H 

        Checking drive:  sg2 RICOHIMG
 DSC_K-1         

        Found camera RICOHIMG
 DSC_K-1         

before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=3, 0, 1, 0)
[S]                                      >>> [F0 24 00 01  00 00 00 00]
[C]                     get_result(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [1C 00 00 00  00 00 01 00]
[R]                              => [1C 00 00 00]
[C]                     read_result(0x3, size=28)
[S]                                      >>> [F0 49 00 00  1C 00 00 00]
[S]                                      <<< [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00 
                                              00 00 00 00  00 00 00 00  00 00 00 00]
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]                                      >>> [F0 4F 00 00  04 00 00 00]
[S]                                      >>> [00 00 00 01]
[C]                     command(fd=3, 0, 0, 4)
[S]                                      >>> [F0 24 00 00  04 00 00 00]
[C]                     get_status(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 00]
[C]             ipslr_status()
[C]                     command(fd=3, 0, 1, 0)
[S]                                      >>> [F0 24 00 01  00 00 00 00]
[C]                     get_result(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [1C 00 00 00  00 00 01 00]
[R]                              => [1C 00 00 00]
[C]                     read_result(0x3, size=28)
[S]                                      >>> [F0 49 00 00  1C 00 00 00]
[S]                                      <<< [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00 
                                              00 00 00 00  00 00 00 00  00 00 00 00]
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=3, 0, 4, 0)
[S]                                      >>> [F0 24 00 04  00 00 00 00]
[C]                     get_result(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [08 00 00 00  00 00 01 00]
[R]                              => [08 00 00 00]
[C]                     read_result(0x3, size=8)
[S]                                      >>> [F0 49 00 00  08 00 00 00]
[S]                                      <<< [92 30 01 00  32 02 00 00]
[... 311 lines cut ...]
raw format                      : DNG
light meter flags               : 3
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 0 (MTF)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.C
af point select                 : Select
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : M-42 or No Lens
battery                         : 8.53V 7.45V 6.67V 0.00V
`

Knubbel, 2016-07-02:
Here is my result using the latest windows exe posted above:

```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  e8 03 00 00 28 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0d 00 00 00 0a 00 00 00  00 00 00 00 01 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 02 00 00 00  02 00 00 00 01 00 00 00
0x0090 | 03 00 00 00 04 00 00 00  06 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 0d 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 02 00 00 00 02 00 00 00  00 00 08 00 00 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  01 00 00 00 07 00 00 00
0x00e0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x00f0 | 01 00 00 00 01 00 00 00  00 00 00 00 88 13 00 00
0x0100 | 04 00 00 00 01 00 00 00  01 00 00 00 00 00 00 00
0x0110 | 01 00 00 00 e8 03 00 00  28 00 00 00 0a 00 00 00
0x0120 | 00 00 00 00 0a 00 00 00  00 00 00 00 0a 00 00 00
[... 176 lines cut ...]
raw format                      : DNG
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 5 (NONE)
user mode flag                  : 2
ae metering mode                : Multi
af mode                         : AF.C
af point select                 : Spot
selected af point               : 524288
focused af point                : 0
drive mode                      : Continuous-HI
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.30
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 5.92V 7.27V 0.00V 0.00V
```

Knubbel, 2016-07-02:
Then I tried taking an picture using the command

> pktriggercord-cli --debug --file_format dng -o test1.dng

It actually did trigger the shutter, but the output in the command line window soon starts rushing by neverendingly. Below is the beginning of it, later it seems to just repeat a couple of lines.

```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  e8 03 00 00 28 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0d 00 00 00 0a 00 00 00  00 00 00 00 01 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 02 00 00 00  02 00 00 00 01 00 00 00
0x0090 | 03 00 00 00 04 00 00 00  06 00 00 00 05 00 00 00
[... 495 lines cut ...]
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 1 res 2
[C]     pslr_buffer_open(#0, type=1, res=2)
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 1 res 2
```

asalamon74, 2016-07-05:
@Knubbel Hmm. Bufmask looks empty.

I suggest to change the camera to bracketing mode (3 pictures, +- 1EV) and check the status reading after that. If the program does not recognize bracketing mode I'll try to fix it.

If the camera recognizes bracketing mode try to take 3 pictures ( --frames 3 ). If it takes 3 pictures and downloads all the pictures after that. If i really takes 3 pictures then the download is buggy. If it takes only 1 picture the exposure is buggy.

Knubbel, 2016-07-08:
I set the camera to Av mode and to 3 image bracketing. This here is what I get for
pktriggercord-cli.exe --status_hex --debug
```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F Generic  STORAGE DEVICE
        Checking drive:  G Generic  STORAGE DEVICE
        Checking drive:  H Generic  STORAGE DEVICE
        Checking drive:  I RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  d0 07 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 01 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  03 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 00 00 00 00  01 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 03 00 00 00  06 00 00 00 04 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 05 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 02 00 00 00 01 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  00 00 00 00 07 00 00 00
0x00e0 | 00 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x00f0 | 01 00 00 00 01 00 00 00  00 00 00 00 88 13 00 00
[... 169 lines cut ...]
raw format                      : DNG
light meter flags               : 128
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 4 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.C
af point select                 : Select
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : on
auto bracket picture count      : 3
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 5.92V 6.92V 0.00V 0.00V
```

Knubbel, 2016-07-08:
Then I triggered it using the same command I used last time.
It did some flickjering three times and then fired three times.
None of the images were saved to sd card.
After firing three times the neverending output in the commandline window started again.
```

pktriggercord-cli.exe --debug --file_format dng -o test2.dng

Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F Generic  STORAGE DEVICE
        Checking drive:  G Generic  STORAGE DEVICE
        Checking drive:  H Generic  STORAGE DEVICE
        Checking drive:  I RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 00
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  d0 07 00 00 0e 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 01 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  03 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 00 00 00 00  01 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 03 00 00 00  06 00 00 00 04 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 05 00 00 00  00 00 00 00 00 00 00 00
[... 1311 lines cut ...]
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 1 res 0
[C]     pslr_buffer_open(#0, type=1, res=0)
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)

```

asalamon74, 2016-07-14:
@Knubbel Sorry for the slow response. The program correctly recognizes the bracketing mode. Could you please add --frames 3 to the last command? Without that the camera only takes one picture even if the bracketing mode is set.

pktriggercord-cli.exe --debug --file_format dng --frames 3 -o test2.dng

Knubbel, 2016-07-17:
Hi.
The camara did shoot 3 images even last time. Now I used your command and it shot 3 times 3 shots, 9 in total. But to be clear here: I already set the camera to 3 shot bracketing mode before pluggin in the USB cable. So that could be the cause.

It is worth mentioning that not a single of the 9 shots is saved on my HDD, it just creates a single zero Byte file named "test3.dng-0000.dng"

And again I had to cancel the command line window as it continued neverendingly.

Here is the output:

```
pktriggercord-cli.exe --debug --file_format dng --frames 3 -o test3.dng
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.01
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F Generic  STORAGE DEVICE
        Checking drive:  G Generic  STORAGE DEVICE
        Checking drive:  H Generic  STORAGE DEVICE
        Checking drive:  I RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  20 03 00 00 2d 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 01 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  03 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
[... 1226 lines cut ...]
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 1 res 0
[C]     pslr_buffer_open(#0, type=1, res=0)
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 1 res 0
[C]     pslr_buffer_open(#0, type=1, res=0)
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)

```

horemachet, 2016-07-17:
Hi,

I did do the test on Linux.
I left the camera settings to default. Means no bracketing settings configured on the camera.

At the beginning the display is off.

Giving 
./pktriggercord-cli --debug --file_format dng --frames 3 -o test2.dng 

the display flashes three times. With the third flash it does one shot.
One file with size 0 is stored to the disk:  test2.dng-0000.dng

I did do the test with and without mounting the sdcards. Makes no difference.

Hope this helps

Karsten

asalamon74, 2016-07-19:
@Knubbel It seems to me Pentax K-1 handles bracketing a different way. On my K-x (after I set bracketing mode) I have to send 3 shutter command to take the three pictures. That's why I suggested --frames 3. K-1 tries to take all the 3 pictures after the first command.

Unfortunately I cannot find the bufmask (the field that shows the status of the camera buffers). And downloading the first buffer saves nothing (it opens the file and waits forever).

My guess: After the shutter command the camera does not save the image into the buffer. I don't know the reason. Maybe someone later will solve it and send a patch.

asalamon74, 2016-07-19:
@horemachet Well at least it shows that it works the same for Linux and Windows, so it's not an OS-specific issue.

horemachet, 2016-09-06:
Any news on this?

asalamon74, 2016-09-08:
@horemachet No

Knubbel, 2016-12-23:
Hi,
I just did a retry using the latest windows build. Generally all seems to be working with the big exception of data transfer. The images are saved neither on card nor HDD. Consequently no preview image is shown in pktriggercord.
I do wonder if tehre is something about the USB connection type.
PKTriggercord only recognizes the camera in MSC mode.
Pentax own Image Transmitter software requires the camera to be in PTP mode.

Here is the output anyway.

```
pktriggercord-cli.exe 0.84.02
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F
        Checking drive:  G RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  0a 00 00 00 50 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  64 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  02 00 00 00 01 00 00 00
[... 172 lines cut ...]
raw format                      : DNG
light meter flags               : 16
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 4 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Select
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 8.35V 6.65V 5.32V 0.00V
```

Knubbel, 2016-12-23:
And here is the result of "**pktriggercord-cli.exe --status_hex --debug --file_format dng -o test1.dng**"

It does look differently from the output weeks ago. For example it doesnt run neverendingly. Is there really no way to locate the buffer?
Or something different in PTP mode?
```
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.02
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F
        Checking drive:  G RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  03 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  0a 00 00 00 50 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  64 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  02 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 05 00 00 00  06 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 05 00 00 00  00 00 00 00 00 00 00 00
[... 273 lines cut ...]
raw format                      : PEF
light meter flags               : 0
ec                              : unknown
custom ev steps                 : 1/2
custom sensitivity steps        : 0
exposure mode                   : 0 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : MF
af point select                 : Auto-5
selected af point               : 0
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : unknown
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : M7B7
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : M-42 or No Lens
battery                         : 0.00V 0.00V 0.00V 0.00V
```

asalamon74, 2016-12-23:
@Knubbel The program only works in MSC mode, not in PTP mode. 

I've created a new test version (based on the bufmask of K-70 see  #14). Please try it.
[pktriggercord-0.84.03-win.zip](https://github.com/asalamon74/pktriggercord/files/670678/pktriggercord-0.84.03-win.zip)

Knubbel, 2016-12-23:
Used your new version, but on CLI it did not do anything:

```
pktriggercord-cli.exe --status_hex --debug --file_format dng -o test1.dng

Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.03
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
pslr.c:418:ipslr_identify(p) failed: 4
1 Cannot connect to Pentax camera. Please start the program as root.

E:\85 Downloads\pktriggercord-0.84.03-win>pktriggercord-cli.exe --status_hex --debug --file_format dng -o test1.dng
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.03
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[... 313 lines cut ...]
raw format                      : DNG
light meter flags               : 16
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 1
exposure mode                   : 4 (NONE)
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Select
selected af point               : 0
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket ev                 : 1.00
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : smc PENTAX-DA 35mm F2.8 Macro Limited
battery                         : 5.77V 6.55V 5.38V 0.00V
```

asalamon74, 2016-12-26:
`--status_hex` just reads the status, it does not take picture. The first output gives s strange error, the second one looks much better. After reading the status the program exists it does nothing.

If you remove `--statux_hex` the program will try to take a picture and download it.

Did you press the shutter on the camera? For a few models you can do that if you add --noshutter instead of --status_hex after that the program waits for your shutter press and downloads the image.

Knubbel, 2016-12-27:
First of all: Merry christmas!

Ok, I now used your latest version 0.84.03 again and tried some more. Now I found a couple of weird things but also some encouraging ones.

Using the GUI version I already knew I could trigger the camera successfully, but never got any image.
But that is not 100% true now.

1a)
When using the GUI version I  checked the "Auto save" box and entered "test1.dng" in the "Name" text box. I also browsed to a folder using the button "Browse..." behind der Text box "Folder".
Now when I trigger the camera it shoots, but nothing ever gets saved anywhere. This is what I reported earlier.
1b)
Now I aggain triggered the camera but then afterwards pushed the "Save As" button in the GUI and now it actually did sucessfully save the shot DNG file!

2a)
Next thing I did is run "pktriggercord-cli.exe --debug --file_format dng --o test1.dng" on the CLI.
A long screen output was created and in the program folder a file named "test1.dng-0000.dng" was created with zero Bytes. So again it did not work to save a DNG.

2b)
Lastly I changed the command to "pktriggercord-cli.exe --debug --file_format jpg --o test1.jpg".
And again to my second suprise it actually did successfully save a JPG file named "test1.jpg-0000.jpg" to the program folder.

Here is the output of that try (which I had to break off since it runs neverendingly):

```
pktriggercord-0.84.03-win>pktriggercord-cli.exe --debug --file_format jpg --o test1.jpg
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.03
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[... 585 lines cut ...]
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA5291540, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA5291540, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA52A1540, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA52A1540, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA52B1540, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA52B1540, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA52C1540, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA52C1540, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
```

Knubbel, 2016-12-27:
More weird stuff. I just wanted to document the 2a) output which I did not capture the first time, so I ran it again. It now did not only create a zero Byte file, but a 14 MByte corrupted file. 14 MByte is obviously not large enough for the 36 MPx. So it did save something. Strange.

Here is the output:

```
pktriggercord-0.84.03-win>pktriggercord-cli.exe --debug --file_format dng --o testraw.dng
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.03
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  03 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  3c 00 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 00 00 00 00  02 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 05 00 00 00  06 00 00 00 05 00 00 00
[... 411 lines cut ...]
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA5698BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA5698BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA56A8BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA56A8BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA56B8BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA56B8BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA56C8BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA56C8BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
```

Knubbel, 2016-12-27:
OK, more testing. I found that the reason for the corrupted DNG file actually was my breaking off of the endlessly scrolling cmd output. Now I let it go on and it did end after a couple of seconds. The result is a working DNG file saved.

Here is the full output:

```
pktriggercord-0.84.03-win>pktriggercord-cli.exe --debug --file_format dng --o testraw2
Debug messaging is now enabled.
pktriggercord-cli.exe 0.84.03
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F RICOHIMG DSC_K-1
        Found camera RICOHIMG DSC_K-1
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=30, 0, 0, 4)
[C]                     get_status(0x30)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=30, 0, 1, 0)
[C]                     get_result(0x30)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x30, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=30, 0, 4, 0)
[C]                     get_result(0x30)
[R]                              => [08 00 00 00]
[C]                     read_result(0x30, size=8)
[R]                              => [92 30 01 00  32 02 00 00]
        id of the camera: 13092
[C]             ipslr_status_full()
[C]                     command(fd=30, 0, 8, 0)
[C]                     get_result(0x30)
[R]                              => [C8 01 00 00]
        read 456 bytes
        expected_bufsize: 456
[C]                     read_result(0x30, size=456)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (424 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  3c 00 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 0a 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 0c 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 00 20 03 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 01 00 00 00 00 00 00 00  02 00 00 00 01 00 00 00
0x0090 | 00 00 00 00 05 00 00 00  06 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  01 00 00 00 01 00 00 00
[... 3652 lines cut ...]
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA7858BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA7868BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA7868BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA7878BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA7878BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA7888BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA7888BB8, 0x10000})
[C]                     command(fd=30, 6, 0, 8)
[C]                     get_status(0x30)
[C]                     get_status(0x30)
[C]     pslr_buffer_read(27F1C0, 65536)
[C]             ipslr_download(address = 0xA7898BB8, length = 65536)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xA7898BB8, 0x1

asalamon74, 2016-12-28:
`--debug` prints out lots of debug messages which makes the program quite slow. Without that it should work faster (still several minutes required for a save).

I'll check autosave, I hardly ever use it.

asalamon74, 2016-12-29:
I tested autosave and it was working correctly for me. For the name part you don't have to add .dng extension. If you enter 'test' as name the program will automatically create test0000.dng, test0001.dng... file names.

I don't really see any K-1 specific reason for a non-working autosave if 'save as' works correctly.

Knubbel, 2016-12-29:
I am still unable to do any autosaves, which is a critical function for me.
All happens under windows 7 64bit. Maybe its an windows version bug? Maybe something to do with path naming conventions?
I generally wonder what the "Auto delete" checkbox is for.
Maybe all this is connected to the GUI never showing any previews?

These are the current issues:

Issue 1: No preview image is ever shown in the GUI

Issue 2: Autofocus points incorrect (still the 11 items, but K-3/K-3II and K-1 have 27

Issue 3: Pressing the "Green Button" Button in the UI in M mode on camera causes the program to infinitely show the clock symbol/freeze

Issue 4: When connecting the GUI version the windows with output says:
(pktriggercord.exe:8776): Gtk-WARNING **: Could not find the icon 'C:\Windows\sy
stem32\GeneIcon.dll,72'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
        http://icon-theme.freedesktop.org/releases
pslr.c:929:ipslr_next_segment(p) failed: 3
Could not get buffer data
pslr.c:929:ipslr_next_segment(p) failed: 3
Could not get buffer data

Issue 5: When the upper GUI checkbox "Auto save" is checked and the lower checkbox "Auto delete" is not checked" and a Name is given as "test" and a folder is given as "E:\downloads" with neither DNMG nor JPG any autosaving happens under windows.

asalamon74, 2016-12-31:
@Knubbel 

Issue 2: The GUI always shows the same 11 items regardless of the camera. Other configurations are not implemented.

Issue 3: Green button does not work for all the cameras.

Issue 4: Right after connection? The program thinks that the camera already has an image in the buffer and tries to show the preview, but it's not possible.

Issue 1 and 5 also shows that no image can be downloaded although the GUI thinks that an image is ready.

All your issues shows that the image cannot be downloaded. Once you mentioned that 'save as' was working, I don't understand how it is possible. It is also very strange that the command line interface was working sometimes.

My best guess is that the bufmask reading is still incorrect.

asalamon74, 2017-06-03:
I new K-1 user started to test pktriggercord, and the results are quite promising so far. He is using the latest firmware ( 1.41.28.16 ), it's possible that upgrading the firmware solves lots of tethering bug.

SM3NSE, 2018-04-03:
Hi.
Is there any progress on the K-1 support? Or is this a dead project?

BR,
Lage

asalamon74, 2018-04-04:
@SM3NSE The test results were promising, but it was not conclusive enough for me to close this issue. Do you have a K-1? Would you like to test it?

SM3NSE, 2018-04-04:
I do have a K-1. How can i contribute? Is there any risk of bricking the camera?

asalamon74, 2018-04-05:
@SM3NSE So far no one informed me about bricking the camera. I've sent unknown commands to my cameras without problems. Sometimes the camera stops and the battery has to be removed and put back.

What is your operating system? If you're using Linux, you can clone the latest version and compile it. If you're using Windows, I can attach a test version.

First, the connection has to be tested using the following command:

`pktriggercord-cli --status_hex --debug`

SM3NSE, 2018-04-05:
I'm on W10. I guess i can setup a linux VM if that is preferrable?

SM3NSE, 2018-04-05:
[MobaXterm_Cmd_20180405_160207.txt](https://github.com/asalamon74/pktriggercord/files/1880319/MobaXterm_Cmd_20180405_160207.txt)
Here you are.

asalamon74, 2018-04-06:
@SM3NSE I've found a bug based on your report and I've fixed it (hopefully). Could you please test the new version: 
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1882893/pktriggercord-0.84.05-win.zip)

SM3NSE, 2018-04-07:
[MobaXterm_Cmd_20180407_220408.txt](https://github.com/asalamon74/pktriggercord/files/1886693/MobaXterm_Cmd_20180407_220408.txt)
yes, it did take a picture, even if the filename came out different to what i expected.

sebas77, 2018-04-07:
Hey is there an android version to test? I am ready to test it

sebas77, 2018-04-07:
Nvm I realized that what I really need is a screen mirroring not a tether app

SM3NSE, 2018-04-08:
[MobaXterm_Cmd_20180408_205306.txt](https://github.com/asalamon74/pktriggercord/files/1887789/MobaXterm_Cmd_20180408_205306.txt)
Tested a 5s and 32s exposure. Both worked.

asalamon74, 2018-04-09:
@SM3NSE That's great.

Could you please try the GUI by starting:

`pktriggercord`

After you take the first image the program should show a thumbnail and a bigger preview. If you try a second image, it should show two thumbails... Please also try to save the image.

SM3NSE, 2018-04-09:
nope, that failed. Cant set user mode. Green button fails. i can trigger camera, but no thumbnails, and it enters bulb mode.
Program crashes after a while if camera is turned off.
[image]
[image]

asalamon74, 2018-04-09:
@SM3NSE 

Green button is not working for newer cameras.

User mode selector is a read-only field, it is not possible to change it.

The 30s shutter speed looks quite bad. What if you first turn on the camera and connect only to the computer later, it solved the same program for an other camera model.

If it's not working, please execute

`pktriggercord --debug`

and attach the output here.

SM3NSE, 2018-04-10:
Nope, still doesn't work properly.
[MobaXterm_Cmd_20180410_122158.txt](https://github.com/asalamon74/pktriggercord/files/1894164/MobaXterm_Cmd_20180410_122158.txt)
[image]

SM3NSE, 2018-04-11:
Hi.
Just realized i wasn't running the latest FW in my K-1. Upgraded from 1.40 to 1.43
Unfortunately, no improved behaviour in pktriggercord...

asalamon74, 2018-04-12:
@SM3NSE I've fixed a bug. Here is a new version: 
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1901783/pktriggercord-0.84.05-win.zip)

SM3NSE, 2018-04-12:
Can't see any difference.
[image]
[MobaXterm_Cmd_20180412_114456.txt](https://github.com/asalamon74/pktriggercord/files/1902449/MobaXterm_Cmd_20180412_114456.txt)

[21 more replies]