asalamon74/pktriggercord #11: Where exactly are we with the K-50?

flipster64, 2016-07-06:
I do have a K-r, and I can use that, but I'm trying to figure out just how much functionality there is for the K-50 there is ... I know it says "bugs".  But I don't know to what extent or what kinds of things we're talking about.

I connected with 0.84 to my K-50 using Windows 10 last night..  I put it in "M" Manual mode.   I could get it to take some pictures.  One thing I noticed it did was set the shutter speed ... sometimes ... to 9999" (that's what the camera UI would show).  The shutter wouldn't stay open 9999 seconds, but it was doing long exposures.  I'd set the PKTrigger UI shutter speed to something else, but it wouldn't change on the camera.

The only way to get it off of this was to re-start the program and adjust from there.  And at some point it did tend to want to go back to that if I dinked with shutter speed.

And another thing it did, twice, was after taking several shots the shutter just started rapid-firing and the only way to stop it was to remove the battery.  

Are these things you've seen?  Or maybe it's a cable problem.  I'm using the one that came with it.  I'm mainly trying to get a feel for what I might expect using it with the K-50 and if these issues are "normal" with the state of the code right now, or if there's something I might tweak.

It's been years since I've programmed, and I must admit I was fairly crappy at OOP, but I get the concept.  

How do you guys go about reverse-engineering what a camera will accept for commands?  I'd help, if I knew how.

asalamon74, 2016-07-08:
I suggest to test the command line version first. You can change shutter speed using −t ( or  −−shutter_speed ). 

Just changing shutter speed:

`pktriggercord-cli -t 1/4 --status`

Does changing works here? You can try different values: 1/100, 1/4, 2, 10 (up to 30).

If it works you can try to take pictures:

`pktriggercord-cli -t 1/4 -o test.dng`

If it still works you can check the source code of the GUI.

You won't need too much OOP for this ;)

farbodab, 2016-09-01:
Sorry for the dumb question but how do you use the pktriggercord-cli on windows. When I open it and try to input commands it won't let me. All I can see is 
[image]

asalamon74, 2016-09-02:
@farbodab Open a console (terminal) windows and type the command you want to use. For instance

`pktriggercord-cli --status`

Blueshawk, 2017-03-11:
Followup for k-50 users. 
I tested my k-50 with the latest code and I was able to get most functions working in the CLI with the exception of bulb without the nasty and scary flutter bug lockup the GUI was doing to them.

It would be great if we got bulb mode working so I could use it for long exposure astrophotography.

@asalamon74 Thank you for all your efforts!

asalamon74, 2017-03-14:
@Blueshawk Thanks for the info. I do close this issue now because a separate issue is already opened for K-50 bulb mode problems: #6

closed 2017-03-14 by asalamon74