asalamon74/pktriggercord #12: Installation & usage for newb please...

gsrokmix, 2016-07-07:
Hi,

I am desperate to be able to tether my K-5 ii to my Macbook Pro. I have read great things about your software by some people on the Pentax forums, and at DP Review. I have downloaded the Windows version, as well as the latest GTK. I am trying to use pktriggercord in a VM on my Mac. I have Windows 7 installed, and am using it through Parallels. I can't for the life of me figure out how to get this software to work. Do I need to go into the command line to get to the GUI.

If you could please help me I'd greatly appreciate it.

Thanks,

George

asalamon74, 2016-07-08:
The program uses quite low-level ways to access the camera. I don't know Parallels but it's quite possible that the virtual computer does not support the same low-level ways to access the camera.

You can start the gui using the pktriggercord command, it should automatically connect to the camera.

If it's not working you may try the command line version:

pktriggercord-cli --status --debug_hex

mccodesurfer, 2017-02-09:
I'm trying to do the same but without VM or bootcamp, but If you have win7 on parallels then you can use pk_tether which works very well with the K-5II.