asalamon74/pktriggercord #13: Doesn't work with K-30

mmxbow, 2016-12-11:
I have a K30 camera with firmware updated to 1.06.
pktriggercord doesn't work wiht this camera. The program cannot set properly a shutter speed and ISO. 
A value of the shutter speed is 9999" in the camera and 0 in status info regardless of a value in settings window. 
[K30_status.txt](https://github.com/asalamon74/pktriggercord/files/644385/K30_status.txt)

asalamon74, 2016-12-13:
Please set the camera to Tv mode and execute the following command:

`pktriggercord-cli --shutter_speed 1/1000 --status_hex`

What is the output? Does it change the shutter speed of the camera?

nikolay-turpitko, 2016-12-15:
Hi! I have a similar problem with my K-30 (firmware v 1.01).

I've tried GUI app and it seems, that it did't change some settings. When I tried to get picture, it opened mirror for several seconds and picture was overbright (it showed 9999" on the camera display). Also I noticed that app showed Av mode, but camera was set to Tv.

I've tried to reset all camera settings to the factory values and tried your command. It seems, that it applied 1/1000 from cli, but didn't apply settings from GUI app. Now it shot with (apparently) 1/1000, at least it showed it on camera display. Also, "focus" button had no effect after settings reset.

I tried to:

1. Unplug USB cord and plug it again.
2. Send  cli command again.
3. Started GUI app.

Shutter speed setting started to apply but with a huge lag. Though, no focus and wrong mode indicator.

asalamon74, 2016-12-21:
Lots of issues here

Strange bugs: The original guy who added K-30 support also reported similar things. Shutter speed changing to 9999s, strange exposure modes. If he was not able to fix these bugs (although he had a K-30), I don't think I'll be able to fix the bugs without a K-30. I've updated [BUGS](https://github.com/asalamon74/pktriggercord/blob/master/BUGS) file.

Firmware: I also added this fact to the BUGS file

Connection: It really sounds strange because my K-x prefers connecting the USB cable first and turning on the camera later. Also added this info to the BUGS file.

GUI: To tell the truth I don't really like (or use) the GUI. The controls show the values read from the camera (updated every 2 seconds) and allows the user to change the values. Maybe someday I'll reorganize the GUI and eliminate this problem. Or hopefully someone else will do it and send me the patch.

nikolay-turpitko, 2016-12-21:
Well, it seems that one can go just so far without official protocol description.

I stopped to play with the program, when instead of performing auto-focus it suddenly cause the camera to zoom in and out so quickly that it jumped on the table and powered off, and won't power on again till I removed battery. Strangely, it executed the same exact command several times before that without an issue... May be timing between commands in sequence matters, don't know.

BTW, how did you crack the protocol? Did you use some software tool to log signals on USB port when executed official program in Windows, or something like that? If this does not require some additional equipment and you can provide some not very complex instruction, I may try to do it myself. I may install Windows XP in the VirtualBox on Ubuntu (if this help), though I have no spare PC to install Windows on it.

asalamon74, 2016-12-24:
Yes, an official documentation would be really helpful Since tethering is not officially supported it is also possible that the firmware has tethering bugs.

My program is a fork of the abandoned [PK-Remote](http://pkremote.sourceforge.net/) probably they used an USB-sniffer originally. 

A few guys also reverse engineered the firmware and collected info [here](https://github.com/asalamon74/pktriggercord/blob/master/pentax_scsi_protocol.md).