asalamon74/pktriggercord #21: integrating with Gphoto2?

Blueshawk, 2017-03-11:
@asalamon74  I have been admiring your work for a long time as I try to set up a remote observatory using my pentax k-50 as the main camera for an Indi/ekos setup. Unfortunately, like many Linux tethering apps, the Indilib drivers use libGphoto2, which in turn uses the outdated Pktether to talk to early pentax cameras and doesn't support the later ones. I approached Gphoto developer @msmeissn but he has his hands full over there with other projects. I've been comparing code between the two and I think reintegrating the much improved pktriggercord to Gphoto would help get the k series tethered in many apps at once. Maybe you guys can get together sometime and being familiar with the code, you might bang something out fairly easily. 

https://github.com/gphoto/libgphoto2

 I'll be glad to help test with my k-50 as time allows. I currently have both programs compiling from current git on my xubuntu (15.10) box as well as my recently updated notebook running xubuntu 16.10

Meanwhile I think I'll poke at that k-50 bulbing problem a bit, it seems more up my street as a repair tech.

msmeissn, 2017-03-13:
I worked a bit on a longer train ride. I think there is similar to the previous pslr port an easy way

- use several pslr* files directly 1:1
- wrapper library.c file that  offers the libgphoto2 driver interface

asalamon74, 2017-03-14:
I think libgphoto2 uses pkremote and not pktether. Pktether is a closed source application, pkremote is the original pentax tethering project I've forked to create pktriggercord.

Integration is OK for me, but I don't plan to modify my code too much. I don't like to idea to add lots of gphoto-specific code to pktriggercord. 

I'll check the pull request soon.

msmeissn, 2017-03-14:
i need to redo it, i was a bit too enthusiastic with the windows part, and travis fails on win32

asalamon74, 2017-03-14:
@msmeissn Yes, I've also realized that. You can check windows compilation using make localwin, that will also download the crosscompiler.

asalamon74, 2017-05-02:
I close this issue, because it's more like an issue for libgphoto2 and not for pktriggercord.

closed 2017-05-02 by asalamon74

ian-morgan99, 2023-09-27:
On this topic, I have raised the question here to ask the makers of libgphoto2 to update their project to reference these libraries, instead of the PKRemote ones, which seem to be badly out of date now. 

I hope that's OK. 

I know this issue was closed, but if there are any clues that can be shared here of what they need to do, it would be greatly appreciated.   Here's the link :   https://github.com/gphoto/libgphoto2/issues/928

msmeissn, 2023-09-28:
commit 3118a779151d9b39b907a2b2817929011b0a9bd3

again, we still need the USB ids of the K3-III and K1-II

ian-morgan99, 2023-09-28:
Thanks for your quick reply.
How do I find those USB id's ? I have both bodies here.
Cheers,
Ian.

asalamon74, 2023-09-28:
I usually find the ids here: https://exiftool.org/TagNames/Pentax.html

ian-morgan99, 2023-09-29:
> commit 3118a779151d9b39b907a2b2817929011b0a9bd3
> 
> again, we still need the USB ids of the K3-III and K1-II

I think I now have them. When I attach the camera, I get the following in the event log. 

I assume you need either:
USB\VID_25FB&PID_0189\8093033

Or maybe:
Vendor IDs:	
		USB\VID_25FB&PID_0189&REV_0200
		USB\VID_25FB&PID_0189
-----------------------------------------------------------------------

I'll now see if I can get the same for the K1ii

A new external device was recognized by the system.

Subject:
	Security ID:		SYSTEM
	Account Name:		DESKTOP-B595CK8$
	Account Domain:		WORKGROUP
	Logon ID:		0x3E7

Device ID:	USB\VID_25FB&PID_0189\8093033

Device Name:	PENTAX K-3 Mark III

Class ID:		{eec5ad98-8080-425f-922a-dabf3de3f69a}

Class Name:	WPD

Vendor IDs:	
		USB\VID_25FB&PID_0189&REV_0200
		USB\VID_25FB&PID_0189
		
		

Compatible IDs:	
		USB\COMPAT_VID_25FB&Class_06&SubClass_01&Prot_01
		USB\COMPAT_VID_25FB&Class_06&SubClass_01
		USB\COMPAT_VID_25FB&Class_06
		USB\Class_06&SubClass_01&Prot_01
		USB\Class_06&SubClass_01
		USB\Class_06
		
		

Location Information:	
		Port_#0001.Hub_#0002

ian-morgan99, 2023-09-29:
Or maybe this is what you need:
A new external device was recognized by the system.

Subject:
	Security ID:		SYSTEM
	Account Name:		DESKTOP-B595CK8$
	Account Domain:		WORKGROUP
	Logon ID:		0x3E7

Device ID:	USB\VID_25FB&PID_0182\7371227

Device Name:	USB Mass Storage Device

Class ID:		{36fc9e60-c465-11cf-8056-444553540000}

Class Name:	USB

Vendor IDs:	
		USB\VID_25FB&PID_0182&REV_0102
		USB\VID_25FB&PID_0182
		
		

Compatible IDs:	
		USB\COMPAT_VID_25FB&Class_08&SubClass_06&Prot_50
		USB\COMPAT_VID_25FB&Class_08&SubClass_06
		USB\COMPAT_VID_25FB&Class_08
		USB\Class_08&SubClass_06&Prot_50
		USB\Class_08&SubClass_06
		USB\Class_08
		
		

Location Information:	
		Port_#0001.Hub_#0002

ian-morgan99, 2023-09-29:
OK. Im summary: Having done those, I think I now understand what you actually needed: [msmeissn](https://github.com/msmeissn)

For the K3iii in MSC mode:
USB\VID_25FB&PID_0189\8093033

For the K1ii in MSC mode:
USB\VID_25FB&PID_0183\7371227
For the K1ii in PcP mode:
USB\VID_25FB&PID_0182\7371227

Please let me know if that's now what you needed. (I think what you actually need is probably in the thread above, if not)

ian-morgan99, 2023-09-29:
I don't think there's much point putting the details of the K3iii in CD-Rom mode, as I think that's only used for software installation.