asalamon74/pktriggercord #23: k10d only gets 288 bytes status?

msmeissn, 2017-03-27:
I bought a K10D and it reports only 288 byte status, instead of 392 bytes.

I adjusted the table, but the ipslr_status_parse_k10d function seems to parse more than 288 bytes?

msmeissn, 2017-03-27:
./pktriggercord-cli --debug --status_hex > k10d.log

[k10d.txt](https://github.com/asalamon74/pktriggercord/files/873576/k10d.txt)

asalamon74, 2017-03-28:
I'm quite sure that 392 is the correct number. Several users reported that pktriggercord handles K10D correctly and the parse function also requires more that 288 bytes.

The official [Pentax Remote Assistant 3](http://www.ricoh-imaging.co.jp/english/support/digital/remoteassistant_k10d_mac.html) requires firmware version 1.20 or later.

What is your firmware version? I've just commited a new feature pktriggercord-cli --read_firmware_version might work for K10D and print out the firmware version. Can you upgrade to 1.20 or later?

Someone already reported this error two years ago, I suggested firmware upgrade but got no response.

msmeissn, 2017-03-28:
yeah, i just bought it used and did not update the firmware (yet).

Firmware version: 1.00.00.07

msmeissn, 2017-03-28:
I updated the firmware to 1.31.00.19 now and the size is now 392 bytes

closed 2017-03-28 by msmeissn

asalamon74, 2017-03-29:
I've added this info to the webpage and to README.md
Thanks for testing.