asalamon74/pktriggercord #24: Please solve all the K-01 Issues.

kamayok3, 2017-04-27:
Hi,

Please solve all the issues in K-01. 

Regards,
Tom

asalamon74, 2017-05-04:
That's quite difficult, because I don't have a K-01. You can help by listing the issues here.

jakeson21, 2017-05-04:
If you kindly send me a K-01, I'll solve all the worlds problems.

asalamon74, 2018-04-24:
@kamayok3 

I've fixed similar bugs for newer camera models, hopefully the same solution also works for K-01.

First, I need the output of the following command twice. Once in the Mode1 you described, once in the mode2.

pktriggercord-cli --settings_hex --debug

kamayok3, 2018-04-25:
Hi,

Below are the result that you need.

**This is with Mode 1:**

C:\Users\Tommy Lim KW\Desktop\pktriggercord-0.84.04-win>pktriggercord-cli --settings_hex --debug
Debug messaging is now enabled.
pktriggercord-cli 0.84.04
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D PENTAX DSC_K-01
        Found camera PENTAX DSC_K-01
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=44, 0, 1, 0)
[C]                     get_result(0x44)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x44, size=28)
[R]                              => [00 00 00 01  00 00 00 00  01 00 01 00  80 00 00 01
                                     00 00 00 00  00 00 00 00  00 00 00 26]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=44, 0, 0, 4)
[C]                     get_status(0x44)
[R]                              => ERROR: 0x00
[C]             ipslr_status()
[C]                     command(fd=44, 0, 1, 0)
[C]                     get_result(0x44)
[R]                              => [1C 00 00 00]
[C]                     read_result(0x44, size=28)
[R]                              => [00 00 00 02  00 00 00 00  01 00 02 10  80 00 00 01
                                     00 00 00 00  00 00 00 00  00 00 00 26]
[C]             ipslr_identify()
[C]                     command(fd=44, 0, 4, 0)
[C]                     get_result(0x44)
[R]                              => [08 00 00 00]
[C]                     read_result(0x44, size=8)
[R]                              => [00 01 2E F8  00 00 02 0B]
        id of the camera: 12ef8
[C]             ipslr_status_full()
[C]                     command(fd=44, 0, 8, 0)
[C]                     get_result(0x44)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0x44, size=452)
[R]                              => [00 00 00 02  00 00 00 00  01 00 02 10  80 00 00 01
                                     00 00 00 00  00 00 00 00  00 00 00 26  00 00 00 00 ... (420 bytes more)]
0x0000 | 00 00 00 02 00 00 00 00  01 00 02 10 80 00 00 01
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 26 00 00 00 00
0x0020 | 00 00 00 0f 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 00 01 00 00 00 00 1e  00 00 00 01 00 00 00 00
0x0040 | 00 00 00 0a 00 00 00 00  00 00 00 0a 00 00 00 00
0x0050 | 00 00 00 0a 00 00 00 0a  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 00 00 00 00  00 00 03 20 00 00 00 64
0x0070 | 00 00 06 40 00 00 00 00  00 00 00 07 00 00 00 07
0x0080 | 00 00 00 01 00 00 00 03  00 00 00 00 00 00 00 01
0x0090 | 00 00 00 00 00 00 00 05  00 00 00 05 00 00 00 05
0x00a0 | 00 00 00 01 00 00 00 01  00 00 00 01 00 00 00 01
0x00b0 | 00 00 00 00 00 00 00 0c  00 00 00 00 00 00 00 01
0x00c0 | 00 00 00 00 00 00 00 02  00 00 00 00 00 00 00 01
0x00d0 | 00 00 00 00 00 00 00 00  00 00 00 07 00 00 00 01
0x00e0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 01
0x00f0 | 00 00 00 01 00 00 00 00  00 00 13 88 00 00 00 03
0x0100 | 00 00 00 01 00 00 00 01  00 00 00 00 00 00 00 01
[... 405 lines cut ...]
0x03c0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x03d0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x03e0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x03f0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00

[C]     pslr_disconnect()
[C]             ipslr_cmd_10_0a(0x0)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x0})
[C]                     command(fd=ac, 10, a, 4)
[C]                     get_status(0xac)
[R]                              => ERROR: 0x00
[C]             ipslr_set_mode(0x0)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x0})
[C]                     command(fd=ac, 0, 0, 4)
[C]                     get_status(0xac)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x00
[C]             ipslr_status()
[C]                     command(fd=ac, 0, 1, 0)
[C]                     get_result(0xac)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xac, size=28)
[R]                              => [00 00 00 01  00 00 00 00  01 00 01 00  80 00 00 01
                                     00 00 00 00  00 00 00 00  00 00 00 26]
[C]     pslr_shutdown()

kamayok3, 2018-04-25:
Because the list is too longggggg.....so, I removed a part of it. Hope this help.

asalamon74, 2018-04-26:
@kamayok3 Thanks, that looks good. I attach a test version:
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1950557/pktriggercord-0.84.05-win.zip)

Please execute the following command:

`pktriggercord-cli --settings`

bulb mode should print press-hold or press-press based on your mode1/mode2 selection.

If that's ok, the next step will be the following:

The solution uses the interval bulb timer, so it will only work if K-01 supports it. (That's why this fix is not working for K-50 as can be seen at issue #6 ).

I need the output of the following command twice. Once when the camera is set to BULB but the timer is not set, second when the camera is set to BULB and the timer is set to a long exposure (preferably 5 minutes). Please also tell me the exact time it was set:

`pktriggercord-cli --settings_hex --debug`

kamayok3, 2018-04-27:
Hi,

in Mode 1:

pktriggercord-cli: K-01 Connected...

one push bracketing             : Unknown
bulb mode                       : press-hold
bulb timer                      : Unknown
bulb timer sec                  : Unknown
using aperture ring             : Unknown
shake reduction                 : Unknown
astrotracer                     : Unknown
astrotracer timer sec           : Unknown
horizon correction              : Unknown
----------------------------------------------------------------------------------------------
in Mode 2:

pktriggercord-cli: K-01 Connected...

one push bracketing             : Unknown
bulb mode                       : press-press
bulb timer                      : Unknown
bulb timer sec                  : Unknown
using aperture ring             : Unknown
shake reduction                 : Unknown
astrotracer                     : Unknown
astrotracer timer sec           : Unknown
horizon correction              : Unknown
-------------------------------------------------------------------------------------------------

Sadly in K-01 doesn't have interval bulb timer option. 

So, any other idea?

Regards,
Tom

asalamon74, 2018-05-01:
@kamayok3 bulb mode reading works correctly.

K-3II has no interval bulb timer option, but it has astrotracer timing, which also works.

If K-01 has no interval bulb timer and no astrotracer then I'm afraid we cannot fix K-01 bulb mode.