asalamon74/pktriggercord #25: Pentax K3-II Bulb mode issue

achellet, 2017-05-14:
Hi !

First : Thanks a lot for your work!

I use Pktriggercord-cli with a raspbian distrib installes on a raspberry pi 3

I observed a bug regarding the bulb mode.
When i launch the following command :

sudo pktriggercord-cli -t 60 -F 3 -d 20 -o /home/pi/Desktop

Pktriggercord-cli displays :

Taking picture 1/3
pslr.c:877:get_status(p->fd) failed: 129
pslr.c:877:get_status(p->fd) failed: 129

I need to press the shutter button to release the shutter.

When the debug mode is activated , i have the following messages :

sudo pktriggercord-cli --debug -t 60 -F 3 -d 20 -o /home/pi/Desktop
Debug messaging is now enabled.
pktriggercord-cli 0.84.04
model (null)
device (null)
[C]     plsr_init()
driveNum:2
        Checking drive:  sg0 RICOHIMG
 DSC_K-3_II

        Found camera RICOHIMG
 DSC_K-3_II

before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=3, 0, 1, 0)
[S]                                      >>> [F0 24 00 01  00 00 00 00]
[C]                     get_result(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [1C 00 00 00  00 00 01 00]
[R]                              => [1C 00 00 00]
[C]                     read_result(0x3, size=28)
[S]                                      >>> [F0 49 00 00  1C 00 00 00]
[S]                                      <<< [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                              00 00 00 00  02 00 00 00  00 00 00 00]
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]                                      >>> [F0 4F 00 00  04 00 00 00]
[S]                                      >>> [00 00 00 01]
[C]                     command(fd=3, 0, 0, 4)
[S]                                      >>> [F0 24 00 00  04 00 00 00]
[C]                     get_status(0x3)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 00]
[... 461 lines cut ...]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 00]
[R]                              => ERROR: 0x00
                shutter result code: 0x0

Thanks for your help!

Blueshawk, 2017-05-15:
Hi achellet, The bulb hasn't been working in the later models because of a change in the firmware. 
 I've been researching the problem with my k-50 and haven't found a good way around it so far. Basically what is happening is the camera is returning an error because the command code for bulb has changed or moved, I suspect after they added the two different bulb types in the extra menu section. 
We also ran into the same problem while updating the Pentax section of Gphoto2
I'm hoping somebody from the original firmware decoding team will pick it up and give us an updated protocol set one day, but I haven't tried to contact them yet.
 It is also possible that the command was broken during that update since Ricoh/Pentax isn't supporting tethering in the lower priced models(grim).

It may be possible to get around this with an external trigger but I haven't tried it yet.

asalamon74, 2017-05-16:
Unfortunately, it's a common problem for newer cameras. I've added this info to the BUGS file.

asalamon74, 2018-02-06:
I've fixed a similar bug for K-70, hopefully the solution will work also for K-3 II. The solution uses the interval bulb timer, so it will only work if K-3II supports it. (That's why this fix is not working for K-50 as can be seen at issue #6 ).

I need the output of the following command twice. Once when the camera is set to BULB but the timer is not set, second when the camera is set to BULB and the timer is set to a long exposure (preferably 5 minutes). Please also tell me the exact time it was set:

`pktriggercord-cli --settings_hex --debug > debug.log 2>&1`

achellet, 2018-02-06:
Hello and thanks a lot for tour work.
I will test it asap and sens you logs and comments.

Envoyé à partir de mon Windows Phone

Nats-ji, 2018-02-06:
> The solution uses the interval bulb timer, so it will only work if K-3II supports it

I think the bulb timer function is only available for K-1, K-70, and KP. But when the GPS is turned on, the timer function in Astrotracer mode can be used on K-3II. I don't know whether this is helpful or not.

Debug log in normal bulb mode with GPS turned on. [debug-notimer.log](https://github.com/asalamon74/pktriggercord/files/1701050/debug-notimer.log)
Debug log in Astrotracer mode with the timer set to 5 minutes. [debug-withtimer.log](https://github.com/asalamon74/pktriggercord/files/1701051/debug-withtimer.log)

[image]

asalamon74, 2018-02-08:
@Nats-ji I can only find one yes/no field difference. Probably it's astrotracer on/off. I'll create a new test version (hopefully tomorrow) which reads this field and I'll also increase the number of fields read by settings_hex, maybe the timer value is stored somewhere else.

achellet, 2018-02-08:
Hello and thanks a lot for your help and work!

This is the log file with and without astrotracer:
[MobaXterm_Pentax_K3_II_20180208_163617.txt](https://github.com/asalamon74/pktriggercord/files/1707769/MobaXterm_Pentax_K3_II_20180208_163617.txt)

asalamon74, 2018-02-09:
@achellet The log only shows that the bulb is not working. You're using a very old version of the program, which does not support --settings and --settings_hex, only the git version supports it.

asalamon74, 2018-02-09:
@Nats-ji @achellet  I've created a new test version. I attach it here:
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1710524/pktriggercord-0.84.05-win.zip)

Hopefully it recognizes the astotracer flag, please test it (turn it on and off) using the following command:

`pktriggercord-cli --settings`

I've increased the settings buffer reading to 2048 bytes, please attach the log (without `--debug` of the following command after setting the astrotracer to 5 minutes:

`pktriggercord-cli --settings_hex`

achellet, 2018-02-10:
Hello !

I've downloaded the test version 084.05 and it works! As you explained i've to set the GPS on and astrotracer off and the exposure time to 60s (using the firmware).The software is not able to set the exposure time, the DSLR seems to have the priority. And  the time the DSLR is ready for a long exposure and start the first capture is a bit long. I've tested the GUI and it works too but seems to not be able to take more than one frame at a time.

[MobaXterm_Pentax_K3_II_20180209_203230_no_debug.zip](https://github.com/asalamon74/pktriggercord/files/1713200/MobaXterm_Pentax_K3_II_20180209_203230_no_debug.zip)

closed 2018-02-10 by achellet

asalamon74, 2018-02-10:
I might find the astrotracer timer field. Could you send me the log after setting the timer to 2 minutes?

`pktriggercord-cli --settings_hex`

Nats-ji, 2018-02-10:
The program crashed on me.
[image]
[pktriggercord-cli --settings --debug.log](https://github.com/asalamon74/pktriggercord/files/1713547/pktriggercord-cli.--settings.--debug.log)
[pktriggercord-cli --settings_hex --debug.log](https://github.com/asalamon74/pktriggercord/files/1713548/pktriggercord-cli.--settings_hex.--debug.log)

achellet, 2018-02-10:
Hi

Here are log files recorded with timer set to 120s and 300s
[MobaXterm_Pentax_K3_II_20180210_203645_timer_120s.txt](https://github.com/asalamon74/pktriggercord/files/1713718/MobaXterm_Pentax_K3_II_20180210_203645_timer_120s.txt)
[MobaXterm_Pentax_K3_II_20180210_204153_timer_300s.txt](https://github.com/asalamon74/pktriggercord/files/1713719/MobaXterm_Pentax_K3_II_20180210_204153_timer_300s.txt)

300s.

asalamon74, 2018-02-11:
@Nats-ji @achellet 

I don't know the reason for the program crash. I've decreased the settings buffer size back to 1024 bytes, hopefully it will work again.

I've found the astrotimer second field, I attach a new version which tries to read it: 
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1714463/pktriggercord-0.84.05-win.zip)

Please test it. If the reading is correct, I'll create a version which changes the setting and hopefully it can be used for better bulb handling.

Nats-ji, 2018-02-11:
It's working now.
[settings_2min.log](https://github.com/asalamon74/pktriggercord/files/1714723/settings_2min.log)
[settings_5min.log](https://github.com/asalamon74/pktriggercord/files/1714724/settings_5min.log)

achellet, 2018-02-12:
Nice work!

I've tested the 0.84.05 version :
[MobaXterm_Pentax_K3_II_20180212_205553_astrotracer_off_300s.txt](https://github.com/asalamon74/pktriggercord/files/1717653/MobaXterm_Pentax_K3_II_20180212_205553_astrotracer_off_300s.txt)
[MobaXterm_Pentax_K3_II_20180212_212445_astrotracer_off_60s.txt](https://github.com/asalamon74/pktriggercord/files/1717654/MobaXterm_Pentax_K3_II_20180212_212445_astrotracer_off_60s.txt)

asalamon74, 2018-02-13:
@Nats-ji @achellet Looks great, I'll create a test version soon which turns on the astotracer and set the seconds.

asalamon74, 2018-02-16:
@Nats-ji @achellet I've created a test version it turns on automatically the astrotracer and sets the seconds. 
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1731827/pktriggercord-0.84.05-win.zip)

The camera has to be in bulb mode. If your camera has different bulb setting it has to be in press-hold (single press bulb mode) and not press-press.

achellet, 2018-02-17:
Hi

Seems to be a nice work. I'll test it and post logs and comment as usual.

Thanks a lot

Envoyé à partir de mon Windows Phone

achellet, 2018-02-17:
Hello.
Here is the log registered with your last test version. It works very well. The exposure time can be set up to 5mn.

I would like to know if the astrotracer is set to off after the expose time is set?

[MobaXterm_Pentax_K3_II_20180217_220355.txt](https://github.com/asalamon74/pktriggercord/files/1734133/MobaXterm_Pentax_K3_II_20180217_220355.txt)

asalamon74, 2018-02-19:
@achellet Right now the program does not change back astrotracer to off, but it's a good idea. I'll modify it.

asalamon74, 2018-02-28:
@achellet Here is a version which changes back astrotracer to off (if it was turned off at the beginning): 
[pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1767524/pktriggercord-0.84.05-win.zip)

Nats-ji, 2018-04-20:
@asalamon74 The timer is working like a charm. But it seems like I'm not able to control the ISO settings.
When I use `pktriggercord-cli -i 100 -t 1 -o test-1 --file_format=dng --debug `, the output file is on ISO1600.
[debug-1.log](https://github.com/asalamon74/pktriggercord/files/1931468/debug-1.log)

And when I use `pktriggercord-cli -i 800 -t 1 -o test-2 --file_format=dng --debug`, the output file is still on ISO 1600.
[debug-2.log](https://github.com/asalamon74/pktriggercord/files/1931469/debug-2.log)

One thing I noticed is that when I set the ISO, the top screen on the camera actually displaying the ISO I set at, but the back screen still showing ISO1600.

asalamon74, 2018-04-22:
@Nats-ji Can you please open a separate issue for the ISO problem? It would be much easier to maintain the issues.