asalamon74/pktriggercord #27: K-500 doesn't work unlike said in page

metal3d, 2017-10-01:
Hello, 
I tried on Fedora 26 (compiled myself) and my pentax k-500 is not found.

That could surprise me that can work while gphoto2 (librairie) cannot use it.

Any chance to find why ?

Output:
```
./pktriggercord-cli --model=K-500 --debug
Debug messaging is now enabled.
./pktriggercord-cli 0.84.04 
model K-500
device (null)
[C]	plsr_init()
driveNum:2
	Checking drive:  sg0 ATA     
 WDC WD10EZEX-22R

	Checking drive:  sg1 ATA     
 WDC WD10EADS-11M

	camera not found
diff: 0.000288
sleep 1 sec
[C]	plsr_init()
driveNum:2
	Checking drive:  sg0 ATA     
 WDC WD10EZEX-22R

	Checking drive:  sg1 ATA     
 WDC WD10EADS-11M

	camera not found
diff: 1.000641
sleep 1 sec

```

Regards :)

asalamon74, 2017-10-02:
A few tips:

- You don't have to specify the model. It is only useful if you connect two (or more models).
- The camera has to be in Mass Storage (MSC) mode, rather than PTP mode.
- You might try executing the command as root.
- I've commited a bugfix related to camera-recognition, if you checkout the latest version from git, it might work. If not it will show more debug messages.

If still not working: What is the device of your mounted camera?

Best,
Andras

asalamon74, 2017-10-16:
@amikot 

What exposure mode do you use? You should use M mode if you want to change both aperture and shutter speed. (ISO setting might not work even in M mode).

Can you see the image in the thumbnails or in the preview after taking the photo?

amikot, 2017-10-20:
Of course I'm using M mode. Yes, I can take picture and I see thumbnail on right side - of course if somehow I will get right exposure (I need to make it in dark room as aperture is always max open and shutter speed is always around 10 sec).

metal3d, 2017-10-23:
Ok the problem was that I need to launch it via sudo...
The preview is very slow BTW, but it seems to work.

I'm now only sad that libgphoto2 doesn't support my K-500 to work with Darktable... but that's another project ;)

I let you close the issue if @amikot problem is resolved too

metal3d, 2017-10-23:
BTW - I also have problems to setup some options, as @amikot

asalamon74, 2017-10-24:
@amikot Thanks for the intensive testing.

About the issues:

-Green and AE lock does not really work on some models.

-ISO changing is not working in M mode (I've tried several ways using my K-x but it was never working)

-Only K-x and K-70 models support --setting.

Current shutter speed 0/0 looks really strange in your debug info. Could you attach the same output twice (in M mode). Once after you set shutter to 1/100 and once after you set to 1/1000. I'll try to locate the field in the hex.

amikot, 2017-10-26:
@asalamon74: Thanks for answer, and your will to do anything for users of Pentax cameras :)

Green and AE Lock is not essential, but ISO and shutter speed would be really good to be able to set.

I just tested this again and just realized that ISO range is wrong for my camera.
It should be 100-51200 but its 100-6400 - maybe it is the reason why its not working (Its not working in all modes, not only in M).
Other thing that is not working is Exposure Correction - changing it does not change anything on camera - so exactly the same as ISO and shutter speed. 

Below are listings of debug after setting shutter to 1/100 and 1/1000. 

```
shutter->1/100
[C]	pslr_set_shutter(1 64)
[C]		ipslr_handle_command_x18(0x16, 2)
[C]		ipslr_cmd_00_09(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=b, 0, 9, 4)
[S]					 >>> [F0 24 00 09  04 00 00 00]
[C]			get_status(0xb)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[R]				 => ERROR: 0x00
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1, 0x64})
[S]					 >>> [F0 4F 00 00  08 00 00 00]
[S]					 >>> [00 00 00 01  00 00 00 64]
[C]			command(fd=b, 18, 16, 8)
[S]					 >>> [F0 24 18 16  08 00 00 00]
[C]			get_status(0xb)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[R]				 => ERROR: 0x00
[C]		ipslr_cmd_00_09(0x2)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x2})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 02]
[C]			command(fd=b, 0, 9, 4)
[S]					 >>> [F0 24 00 09  04 00 00 00]
[C]			get_status(0xb)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[... 176 lines cut ...]
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[R]				 => ERROR: 0x00
```

asalamon74, 2017-10-28:
@amikot Which version do you use? I've already fixed an ISO range bug similar to yours.

amikot, 2017-10-28:
@asalamon74: 

`https://planefinder.net/data/aircraft/M-OVIE`

asalamon74, 2017-10-29:
@amikot I've fixed the ISO range bug, now it should show you the range 100-51200. To tell you the truth, I don't really think it will make ISO changing possible.

To debug you sent me looks OK, it seems to me the camera just ignores the command.
Could you please manually set the camera to 1/100 (in M mode) and attach the output of pktriggercord-cli --status_hex --debug . And the same for 1/1000.

amikot, 2017-10-29:
After doing first of tests that you asked for, changing shutter speed started to work, but aperture stopped, but with next try, everything except ISO was working.

I made some more tests and found out that:
If I will plug cable into the camera first, and launch pktriggercord and than I will switch the camera on (as I was doing always), I will get 0 shutter speed error, and shutter speed will be locked locked. 
But If I will switch the camera on first and than I plug it to the computer, than everything will work as should.

For now problem is solved (except ISO and EV), so I abandoned posting debug results.

I noticed few minor things that seems to be wrong:
1.  ISO range is not the same in all modes. Only in M I can see full range for my camera - in other modes its limited to 6400. It is happening even on User modes 1 and 2 - when M is selected manually on program side.
So best option for now Is to use M - selected on camera.
2. GUI changes look depending of mode and sometimes looks really strange.
3. Settings window doesn't remember picture download folder and filename (so every time when starting I need to set it again).
4. Sliders ad mechanism for setting shutter speed and other options is not good. It would be better to use selectors (like for resolution and quality) or even text fields (I saw you already used that in one of manually selected modes - when was User Mode). Sliders are having big lag and even small move is applied to camera, and sometimes changing values to other than wish to.

BTW. I saw on their page that Gphoto2 uses pktriggercord routines, but it still not working for me. When trying to take photo witg gphoto2 - getting error that its not supported by camera. That's small off topic.

asalamon74, 2017-11-01:
@amikot 

* This cable connection solution is very interesting. I'll also check it using my camera(s), hopefully it will eliminate some bugs.
* I should create some troubleshooting file to list these tricks.
* ISO range bugfix: I've commited a change a few days ago. The program has 3 different ISO tables (one for 1/3 EV, one for 1/2EV, one for 1EV) and some of the tables were incorrect. I've checked the code again, and I don't recognize any more errors. I can add debug messages to the ISO range selection if you are ready to test it.
* Could you show screenshots for the strange GUI change? The GUI changes sometimes, for instance for BULB mode it's a bit different.
* To tell you the truth I also don't like the sliders. I'll change them sooner or later.
* Maybe gphoto2 uses and older version of pktriggercord.

amikot, 2017-11-02:
@asalamon74 
I still didn't get this version where you fixed ISO range. So Its probably fixed - may be no need to do any other tests.
Strange GUI I saw only once - when was doing tests last time - Selectors of resolution, quality and image tone - these were very wide, and all labels were tripled. So there was for example ***  ***  ***  as quality instead of ***.

asalamon74, 2017-11-07:
@amikot 

I've created a [tips and tricks](http://pktriggercord.melda.info/tips_and_tricks.html) file which lists your connection order trick.

Strange GUI: Probably something went wrong during camera-specific initialization. It is quite difficult to debug it if it's not possible to reproduce it. I've cleaned up the GTK code a bit.

The program continuously reads status info from the camera and updates the controls.  Users can send changes to the camera using the exact same controls. That is the root of the problem. Using an apply button would not help here. If you change the ISO but don't send the ISO-change request to the camera, the GUI component will be updated in the next cycle before you can press the apply button.

amikot, 2017-11-12:
@asalamon74 
Its really good idea to make tips and tricks list.
Strange GUI never happened again, so it was minor thing.
I understand how it works, but I just don't understand why you prefer to updates continuously instead of "on demand".
Finally it doesn't matter - because gui version is more like feature test program. Cli version is more useful. Can be used as script command - for example in timelapse script - where number of photos would excess size of SD card.

asalamon74, 2017-11-14:
@amikot I close this issue because the K-500 specific problems are (more or less) fixed.

To tell you the truth also also like the command line interface better than the GUI.

closed 2017-11-14 by asalamon74