asalamon74/pktriggercord #28: ioctl: invalid argument on trying to connect to a K-50

tostiheld, 2017-10-02:
Hi,

First of all, I thought tethering wasn't supported for my model, until I found your tool. Thanks so much for this! Secondly..: sadly it isn't working. I'd be happy to help fix this bug. The full debug log is:

```
thomas@tosti-ijzer ~ $ sudo pktriggercord-cli --debug
Debug messaging is now enabled.
pktriggercord-cli 0.84.04 
model (null)
device (null)
[C]	plsr_init()
driveNum:4
	Checking drive:  sg2 HL-DT-ST
 DVD+-RW GS30N   

	Checking drive:  sg0 ATA     
 KINGSTON SV300S3

	Checking drive:  sg3 PENTAX  
 DSC_K-50        

	Found camera PENTAX  
 DSC_K-50        

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
ioctl: Invalid argument
pslr.c:1558:scsi_write(fd, cmd, sizeof (cmd), 0, 0) failed: 1
pslr.c:1204:command(p->fd, 0, 1, 0) failed: 1
pslr.c:429:ipslr_status(p, statusbuf) failed: 1
1 Cannot connect to Pentax camera. Please start the program as root.
```

tostiheld, 2017-10-02:
So after some printing of the errno variable in the code, I found out that ioctl sets errno to 22 (EINVAL), which it states, but I was to stubborn. Apparently it means request or argp is invalid (ioctl parameters).

Blueshawk, 2017-10-02:
Hi Thomas, I'm a fellow K-50 owner and suffer and got myself involved in trying to tether it earlier this year. Some developments have been made with both pktriggercord and gphoto2, which is also worth a try. See libgphoto2 issue 31 for development notes on that end. 
  Be sure your system doesn't automount the card and be sure to set the k-50 to MSC mode in menus. The default is PTP and while it _should_ have been the standard, Pentax decided not to use it. You will also find that using the dial and menus to set things up works better. Despite our collective best efforts, Bulb still doesn't work on this model, though you might get some results using an external trigger. Beware of the dreaded mirror flop bug if you try that and be ready to pull the battery if your camera suddenly sounds like a bird got stuck in it. Also, avoid the GUI as it's been really unstable and caused the bug to appear in mine.  Hope this helps!