asalamon74/pktriggercord #30: Issue with K3II

Nats-ji, 2018-01-29:
The preview won't show up after the camera was connected. And the program froze up after clicking the take picture button. I think the problem is caused by the updated firmware. The firmware version is v1.10. 

There is the debug log:

```
C:\Users\Nats_\Desktop\pktriggercord-0.84.02-win>pktriggercord-cli --debug
Debug messaging is now enabled.
pktriggercord-cli 0.84.02
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D RICOHIMG DSC_K-3_II
        Found camera RICOHIMG DSC_K-3_II
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=c8, 0, 1, 0)
[C]                     get_result(0xc8)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xc8, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  00 00 00 00
                                     00 00 00 00  03 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=c8, 0, 0, 4)
[C]                     get_status(0xc8)
[R]                              => ERROR: 0x01
[C]             ipslr_status()
[C]                     command(fd=c8, 0, 1, 0)
[C]                     get_result(0xc8)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xc8, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=c8, 0, 4, 0)
[C]                     get_result(0xc8)
[R]                              => [08 00 00 00]
[C]                     read_result(0xc8, size=8)
[R]                              => [9C 30 01 00  3A 02 00 00]
        id of the camera: 1309c
[C]             ipslr_status_full()
[C]                     command(fd=c8, 0, 8, 0)
[C]                     get_result(0xc8)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0xc8, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (420 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 0b 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  7d 00 00 00 28 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 05 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 00 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 80 0c 00 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x0090 | 01 00 00 00 04 00 00 00  05 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 03 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 03 00 00 00 06 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 05 00 00 00 00 00 00 00  06 00 00 00 01 00 00 00
0x00e0 | 01 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x00f0 | 01 00 00 00 00 00 00 00  88 13 00 00 04 00 00 00
0x0100 | 00 00 00 00 00 00 00 00  00 00 00 00 02 00 00 00
0x0110 | 01 00 00 00 28 00 00 00  0a 00 00 00 00 00 00 00
[... 968 lines cut ...]
[C]                     command(fd=c8, 0, 8, 0)
[C]                     get_result(0xc8)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0xc8, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (420 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
get buffer 0 type 2 res 0
[C]     pslr_buffer_open(#0, type=2, res=0)
[C]             ipslr_status_full()
[C]                     command(fd=c8, 0, 8, 0)
[C]                     get_result(0xc8)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0xc8, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00  00 00 00 00 ... (420 bytes more)]
        p->status.bufmask = 0
        No buffer data (0)
^C
```

asalamon74, 2018-01-29:
It seems to me, you're using and old version 0.84.02, could you please try the latest release 0.84.04?

Nats-ji, 2018-01-29:
I tried the latest release 0.84.04 before 0.84.02.
A pentax forum member posted pktriggercord was working with his K-3II (firmware v1.0), that was 0.84.02. So I tried that version with my K-3II (firmware v.1.10), but it didn't work either.
Just in case you need the 0.84.04 debug log:
```
C:\Users\Nats_\Desktop\pktriggercord-0.84.04-win>pktriggercord-cli --debug
Debug messaging is now enabled.
pktriggercord-cli 0.84.04
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D RICOHIMG DSC_K-3_II
        Found camera RICOHIMG DSC_K-3_II
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=c4, 0, 1, 0)
[C]                     get_result(0xc4)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xc4, size=28)
[R]                              => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=c4, 0, 0, 4)
[C]                     get_status(0xc4)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x00
[C]             ipslr_status()
[C]                     command(fd=c4, 0, 1, 0)
[C]                     get_result(0xc4)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xc4, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=c4, 0, 4, 0)
[C]                     get_result(0xc4)
[R]                              => [08 00 00 00]
[C]                     read_result(0xc4, size=8)
[R]                              => [9C 30 01 00  3A 02 00 00]
        id of the camera: 1309c
[C]             ipslr_status_full()
[C]                     command(fd=c4, 0, 8, 0)
[C]                     get_result(0xc4)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0xc4, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 ... (420 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 00 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0020 | 0b 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  7d 00 00 00 1c 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 05 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 00 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 80 0c 00 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x0090 | 01 00 00 00 04 00 00 00  05 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 03 00 00 00  00 00 00 00 00 00 00 00
0x00c0 | 03 00 00 00 06 00 00 00  00 40 00 00 00 00 00 00
0x00d0 | 05 00 00 00 00 00 00 00  06 00 00 00 01 00 00 00
0x00e0 | 01 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x00f0 | 01 00 00 00 00 00 00 00  88 13 00 00 04 00 00 00
0x0100 | 00 00 00 00 00 00 00 00  00 00 00 00 07 00 00 00
0x0110 | 0a 00 00 00 1c 00 00 00  0a 00 00 00 00 00 00 00
[... 255 lines cut ...]
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
^C
```

asalamon74, 2018-02-02:
Your first debug info might indicate that the buffer mask reading is not correct. I've created a test version to use a different (hopefully more reliable) way to read this info: [pktriggercord-0.84.05-win.zip](https://github.com/asalamon74/pktriggercord/files/1689390/pktriggercord-0.84.05-win.zip)

Unfortunately it is also possible that tethering works for one firmware version and not for an other firmware version.

Nats-ji, 2018-02-02:
Thanks for the reply. Here are the new debug logs. By the way, is it possible to save the image to the memory card instead of reading the buffer from the camera and download it to the computer?

```
PS C:\Users\Saime\Desktop\pktriggercord-0.84.05-win> .\pktriggercord-cli --debug
Debug messaging is now enabled.
command line:
C:\Users\Saime\Desktop\pktriggercord-0.84.05-win\pktriggercord-cli.exe --debug
C:\Users\Saime\Desktop\pktriggercord-0.84.05-win\pktriggercord-cli.exe 0.84.05
model (null)
device (null)
[C]     plsr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F
        Checking drive:  G
        Checking drive:  H
        Checking drive:  I
        Checking drive:  J RICOHIMG DSC_K-3_II
        Found camera RICOHIMG DSC_K-3_II
before connect
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=d4, 0, 1, 0)
[C]                     get_result(0xd4)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xd4, size=28)
[R]                              => [02 00 00 00  00 00 00 00  00 02 00 02  01 00 00 80
                                     00 00 00 00  02 00 00 00  00 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[C]                     command(fd=d4, 0, 0, 4)
[C]                     get_status(0xd4)
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x01
[R]                              => ERROR: 0x00
[C]             ipslr_status()
[C]                     command(fd=d4, 0, 1, 0)
[C]                     get_result(0xd4)
[R]                              => [1C 00 00 00]
[C]                     read_result(0xd4, size=28)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=d4, 0, 4, 0)
[C]                     get_result(0xd4)
[R]                              => [08 00 00 00]
[C]                     read_result(0xd4, size=8)
[R]                              => [9C 30 01 00  3A 02 00 00]
        id of the camera: 1309c
[C]             ipslr_status_full()
[C]                     command(fd=d4, 0, 8, 0)
[C]                     get_result(0xd4)
[R]                              => [C4 01 00 00]
        read 452 bytes
        expected_bufsize: 452
[C]                     read_result(0xd4, size=452)
[R]                              => [02 00 00 00  00 00 00 00  10 02 00 01  01 00 00 80
                                     00 00 00 00  00 00 00 00  00 00 00 00  3F 00 00 00 ... (420 bytes more)]
0x0000 | 02 00 00 00 00 00 00 00  10 02 00 01 01 00 00 80
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 3f 00 00 00
0x0020 | 0b 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 01 00 00 00  7d 00 00 00 28 00 00 00
0x0040 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 05 00 00 00 0a 00 00 00  00 00 00 00 01 00 00 00
0x0060 | 00 00 00 00 00 00 00 00  00 00 00 00 64 00 00 00
0x0070 | 80 0c 00 00 00 00 00 00  07 00 00 00 07 00 00 00
[... 1551 lines cut ...]
[C]                     command(fd=d4, 20, 9, 4)
[C]                     get_result(0xd4)
[R]                              => [04 00 00 00]
[C]             ipslr_read_setting() bytes: 4
[C]                     read_result(0xd4, size=4)
[R]                              => [FF 00 00 00]
[C]             ipslr_read_setting(171)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xAB})
[C]                     command(fd=d4, 20, 9, 4)
[C]                     get_result(0xd4)
[R]                              => [04 00 00 00]
[C]             ipslr_read_setting() bytes: 4
[C]                     read_result(0xd4, size=4)
[R]                              => [00 00 00 00]
[C]             ipslr_read_setting(172)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xAC})
[C]                     command(fd=d4, 20, 9, 4)
[C]                     get_result(0xd4)
[R]                              => [04 00 00 00]
[C]             ipslr_read_setting() bytes: 4
[C]                     read_result(0xd4, size=4)
[R]                              => [02 00 00 00]
[C]             ipslr_read_setting(173)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0xAD})
```

asalamon74, 2018-02-04:
@Nats-ji Well, this debug log looks ok to me, the program was reading the settings, maybe you stopped it be too early (`--debug` adds lots of debug messages and makes it very slow).

By default the program writes the image to the standard output (which is not a good idea, I'll change that), could you test it by specifying an output file?

`pktriggercord-cli -o test.dng --file_format=dng --debug
`

Nats-ji, 2018-02-04:
It worked. I got the dng file.

[debug.log](https://github.com/asalamon74/pktriggercord/files/1692880/debug.log)

asalamon74, 2018-02-05:
@Nats-ji 

It is not possible the save the image to the sd card. 

I've tested the program using the standard output writing, and it was working for me. Different operating system (Linux), different camera (K-70). Maybe it does not work on Windows.

I was not a big fan of the default mode of the program, but I avoided the change because of compatibility reasons. Now that I see that it causes more problems, I've modified the code, from now on the program prints out the status info if no other action has been specified. It is still possible to write to the standard output if someone really wants that (`-o -`).

I rename this issue to include the "standard output" string.

Thanks for reporting and testing.

closed 2018-02-05 by asalamon74

Nats-ji, 2018-10-23:
I thought the output file written problem was fixed back in February. 
Turns out it wasn't that simple. At that time, I tested the program with a manual lens attached, and I got the output file. 
However, the time when I opened this issue, I was having an AF lens attached. 
Recently I figured out that as long as I have an AF lens attached, I wouldn't get any output file.