asalamon74/pktriggercord #31: Issue with K20d

seeschloss, 2018-02-04:
I'm trying to use this with my K20d (firmware 1.04) but it seems stuck somewhere, I'm attaching the output of `pktriggercord-cli --debug --status_hex` here: [k20d.txt](https://github.com/asalamon74/pktriggercord/files/1692680/k20d.txt).

I've tried to understand a bit what was going on, and... well according to [the protocol](https://github.com/asalamon74/pktriggercord/blob/master/pentax_scsi_protocol.md) the answer my camera gives (`[00 00 00 00  00 00 00 `**`84`**`]`) which means it's not ready, and a status code that is not supposed to happen.

And I have no idea what I might be doing wrong.

asalamon74, 2018-02-04:
@seeschloss K20D support was added a long time ago, but it would still be better to use a newer version instead of the old 0.84.00, debug messages has been improved a bit.

You might also check [http://pktriggercord.melda.info/tips_and_tricks.html](http://pktriggercord.melda.info/tips_and_tricks.html) for a few tips.

seeschloss, 2018-02-04:
Thanks, it's the same with the git version though ([k20d.txt](https://github.com/asalamon74/pktriggercord/files/1692769/k20d.txt)).

The camera is set to "PC" which is supposed to be UMS I guess, my privileges are fine, the firmware is up to date, I don't use liveview, I have not mounted or unmounted the SD card, and the rest of the points don't seem to apply here.

Actually, if I put the camera in liveview and run pktriggercord, it gets out of liveview (nothing else happens afterwards) so clearly there is some kind of connection that gets established.

asalamon74, 2018-02-06:
@seeschloss 

The new debug log also shows version 0.84.00 (maybe you accidentally attached the same file again).

According to the log the program was able to connect to the camera, read the identification info (recognizing K-20D) but it is unable to read the status information. As you correctly guessed `00 84` is not good, it should be `01 00`.

Quite a few users tested K-20D successfully, so I have no idea. Probably some kind of local problem (bad cable?).

seeschloss, 2018-02-08:
Ah yeah I must have messed up my log (the real one is here 
[k20d-2.txt](https://github.com/asalamon74/pktriggercord/files/1706789/k20d-2.txt)) though I don't think that it helps more.

I'm going to go with the bad cable hypothesis because as I understand, the camera should display "PC" or something when it's plugged on a computer, and mine doesn't. Also, it shows up as /dev/sdb but with "empty media" and I can't access the files from there, so something must be off somewhere. I'm ordering a new cable and closing this issue, thanks for taking the time to think about it!

closed 2018-02-08 by seeschloss

asalamon74, 2018-02-08:
@seeschloss Yes, this log looks better, but does not really help.