asalamon74/pktriggercord #34: Support for K-1 II

hatschepsut, 2018-06-12:
Hi Folks, 

I have just tested with my Pentax K-1 II.
Any time I try to connect it says: "Unknown Pentax camera. 1 Unknown Pentax camera found."

It does work with my K-3 II, though. Hence I assume it is not a general issue with installation or similar.

Are there any plans to add K-1 II to the supported models?

Kind Regards,

Karsten

asalamon74, 2018-06-14:
K-1 II is not yet supported. If you're ready to test, I can create a test version. What operating system do you use?

maydn, 2018-08-19:
I'd also like to see support for the K-1 II, if you need a tester I can try it out. Linux would be preferable, and am happy to compile myself if needed.

asalamon74, 2018-08-19:
Thanks @maydn for testing K-1II. I've added support for K-1 II, basically I just assume that it works like K-1 (which still has limitatations). If you checkout the latest code from git, you can compile it.

You can find a documentation about the basic testing here: [http://pktriggercord.melda.info/debug.html](http://pktriggercord.melda.info/debug.html). Just start reading the status info and please copy the debug here.

maydn, 2018-08-19:
Yeah the K-1 II uses the same mainboard (and is basically the same camera with an IAU chip added).

It's being picked up straight away with the new build, here's the full debug output: https://gist.github.com/maydn/08bf0f79013b1fb42dd9861cb39b985a

maydn, 2018-08-19:
I've been doing a bit of testing with functions that work and trying to compare to the K-1 issue thread to see what is currently working and what is not.

So far, ISO, aperture, shutter speed (on manual mode) and quality all seem to work as expected.

Debug output for command:
`pktriggercord-cli -o test.dng --file_format=dng --debug`
https://gist.github.com/maydn/2f8eb7d034514fb4b9fc6d046da4aad0

I am however, having issues with bulb mode and any shutter speed over 30 seconds, which seems to either lock the firmware forcing a hard restart (I have to disconnect and reconnect the battery) to get the camera working again or just ignores my shutter speed setting.

I get the error "New bulb mode is not supported for this camera model". Using the GUI gives the same error, though it doesn't try to take a picture so doesn't cause crashing.

Debug output for command:
`pktriggercord-cli -o test.dng --file_format=dng --shutter_speed=0.5 --debug`
https://gist.github.com/maydn/f644ae499098b8ff3ca7f1b7a753922c

Debug output for command (which ignored my shutter speed):
` pktriggercord-cli -o test.dng --file_format=dng --shutter_speed=32 --debug`
https://gist.github.com/maydn/aaed68dc0d82435a431adc2f322c4cb1

Hopefully this is helpful to you.

asalamon74, 2018-08-20:
@maydn Bulb mode was under development for K-1, I've commited the same code for K-1 II, if you recompile the program you might try this for K-1 II. It might fix the problem.

asalamon74, 2019-04-22:
Anyone here with a K-1 II who is ready to test the latest version (0.85.00)? I'm specially interested in the bulb mode, but any other info would be useful.

ian-morgan99, 2022-06-06:
I just found this project . I have a K01, K3iii and K1ii and would be happy to test all if it helps the project .

asalamon74, 2022-06-12:
K-01: It has been tested but has lots of bugs. See: https://github.com/asalamon74/pktriggercord/issues/24
K-1 II: As you see in this issue, it has some limitations. It would be great if you could test it and summarize the results here.
K-3 III: Not yet tested, we can continue this in #74

ian-morgan99, 2022-06-12:
Do you have a standard test script to go through please, so my testing is not too 'ad-hoc' ?

asalamon74, 2022-08-07:
@ian-morgan99 It is best to start with reading the status info. If that works, taking a picture. Commands can be found here:

https://pktriggercord.melda.info/debug.html

ian-morgan99, 2023-01-15:
I finally got a chance to test:

This is the result of : Reading status info
pktriggercord-cli --statux_hex --debug

[debug-statusinfo.txt](https://github.com/asalamon74/pktriggercord/files/10421304/debug-statusinfo.txt)

ian-morgan99, 2023-01-15:
Result of 'take a picture' when in 'Manual' mode:

pktriggercord-cli -o test.dng --file_format=dng --debug

[takepicture.txt](https://github.com/asalamon74/pktriggercord/files/10421329/takepicture.txt)

A DNG file was produced.  (I can't upload it as it's an unsupported file type, so I've put it in a ZIP):

[test-0000.zip](https://github.com/asalamon74/pktriggercord/files/10421331/test-0000.zip)