asalamon74/pktriggercord #35: K3II can't read output file with AF lens

Nats-ji, 2018-10-23:
This is a followup issue to [Issue with K3II - Writing to standard output](https://github.com/asalamon74/pktriggercord/issues/30).

This problem is that I thought the output file written problem was fixed back in February.
Turns out it wasn't that simple. At that time, I tested the program with a manual lens attached, and I got the output file.
However, at the time when I opened that issue, I was having an AF lens attached.
Recently I figured out that as long as I have an AF lens attached, I still wouldn't get any output file.

Here are some debug logs when I have AF lens and manual lens attached.
[debug with M lens.log](https://github.com/asalamon74/pktriggercord/files/2507432/debug.with.M.lens.log)
[debug with AF lens.log](https://github.com/asalamon74/pktriggercord/files/2507433/debug.with.AF.lens.log)

asalamon74, 2018-12-21:
I don't really see any reason for behaving differently for M and AF lenses. Maybe it's not able recognise the AF lens? You can check it with `--status`. Or maybe it tries to autofucus and that's the problem?

Nats-ji, 2020-07-13:
I don't think it was trying to autofocus because the shutter was already released. And I think the lens can be recognized correctly. 
Here are the debug logs using the lastest release. `pktriggercord-cli.exe -i 800 -a 10 -t 1/30 --file_format dng -o test.dng --debug > debug.log 2>&1`
[debug_with_lens.log](https://github.com/asalamon74/pktriggercord/files/4910490/debug_with_lens.log)
[debug_without_lens.log](https://github.com/asalamon74/pktriggercord/files/4910493/debug_without_lens.log)

```
pktriggercord-cli.exe: K-3II Connected...

current iso                     : 800
current shutter speed           : 1/30
camera max shutter speed        : 1/8000
current aperture                : 10.0
lens max aperture               : 22.0
lens min aperture               : 2.8
set shutter speed               : 1/30
set aperture                    : 10.0
fixed iso                       : 800
auto iso                        : 100-3200
jpeg quality                    : 4
jpeg resolution                 : 24M
jpeg image tone                 : Natural
jpeg saturation                 : 4
jpeg contrast                   : 4
jpeg sharpness                  : 4
jpeg hue                        : 4
zoom                            : 16.00 mm
focus                           : 20
color space                     : sRGB
image format                    : RAW
raw format                      : DNG
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3 EV steps
custom sensitivity steps        : As EV steps
exposure mode                   : 6
scene mode                      : NONE
user mode flag                  : 0
ae metering mode                : Multi
af mode                         : MF
af point select                 : Unknown value: 6
selected af point               : 8388608
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket picture counter    : 0
auto bracket ev                 : 1.70
shake reduction                 : on
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : -3.30
lens                            : smc PENTAX-DA* 16-50mm F2.8 ED AL [IF] SDM
battery                         : 6.46V 6.08V 0.00V 0.00V
buffer mask                     : 0000000000000000
```

Nats-ji, 2020-07-13:
I think I've just found out what exactly causing this issue. It's the `Lens Correction` settings on the camera. Now it's working like a charm. Maybe you should add a footnote in the documentation to let people know they should turn off the `Lens Correction` on their cameras.
However, I can't seem to control the ISO settings. No matter what ISO I set to the output image's ISO is always the ISO set on camera before connecting to the computer. I will open a new post for this issue.

closed 2020-07-14 by Nats-ji