asalamon74/pktriggercord #36: Support for Samsung GX-1L

tempUser99, 2018-12-20:
I am looking at a GX-1L and attempting to get it up and running.

What can I do to help out and get support for this unit?

I have tried a few things and found the first issue begins when attempting to `get_drive_info` from `pslr_scsi_linux.c`. It appears lines 138-144 are when the check fails on the GX-1L.

If we are able to pass those lines successfully it does appears the camera with change from 'PC' on the LED to the normal readout.

Let me know what I can do to help contribute.

asalamon74, 2018-12-21:
Samsung GX-1L is basically Pentax *ist DL2, but none of them is supported right know.

Adding a new camera requires to add a new line to the [camera_models](https://github.com/asalamon74/pktriggercord/blob/master/pslr_model.c#L935) array in `pslr_model.c`. (Later a camera model specific parser could be also very useful).

I would start be copying the `*ist D` line and replace the name and the id. Ids can be found here: https://www.sno.phy.queensu.ca/~phil/exiftool/TagNames/Pentax.html

Unfortunately `*ist` models are not really working, someone added the support a long time ago (maybe it was working that time, maybe it was only a test), but no one reported me a successful test after that.

It would be really great if you can add the support.

tempUser99, 2018-12-21:
It actually looks like the GX-1L has been added to [camera_models](https://github.com/asalamon74/pktriggercord/blob/master/pslr_model.c#L935).

Unfortunately right now I am seeing issues right at the beginning of the calls. I am attempting to diagnose the issue some but it does not appear to be responding to SCSI calls at all. When testing the response is consistently blank.

Any other thoughts on getting any response? The really interesting part is that even though I am getting no response when the LCD changes it tends to make me feel like the camera is hearing something just not saying anything back at this point.

Is there a simple way to make very simple calls one by one with some debug information? I am thinking we might be able to find some limited functionality but other calls for status aren't getting anything back.

tempUser99, 2018-12-22:
Does the following output help determine anything? I am unsure exactly what calls are made.

pi@raspberrypi:~/Desktop/pktriggercord_PS $ sudo ./pktriggercord-cli --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli --debug 
./pktriggercord-cli 0.84.05 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
	Checking drive:  sg0 Samsung 
 Digital Camera  

Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 40  03 00 00 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 40  03 00 00 04]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 00 00 00]
[C]			read_result(0x3, size=8)
[S]					 >>> [F0 49 00 00  08 00 00 00]
[S]					 <<< []
[R]				 => [00 01 2B 80  00 00 1D F3]
	id of the camera: 12b80
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
[... 141 lines cut ...]
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
^C

asalamon74, 2018-12-22:
Yes, I missed the GX-1L line. I've checked my e-mails,we tried to add support in 2015, but it was not working.

Your log shows that the camera is recognized, it's even possible to read some info (like firmware version).

Instead of taking a picture, let's start with status reading. What is the debug output of the following command:

`pktriggercord-cli --statux_hex --debug`

tempUser99, 2018-12-22:
Below is what we have when running `pktriggercord-cli --status_hex --debug`. (You meant status not statux correct?)

pi@raspberrypi:~/Desktop/pktriggercord_PS $ sudo ./pktriggercord-cli --status_hex --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli --status_hex --debug 
./pktriggercord-cli 0.84.05 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
	Checking drive:  sg0 Samsung 
 Digital Camera  

	Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 00 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 04]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 00 00 00]
[C]			read_result(0x3, size=8)
[S]					 >>> [F0 49 00 00  08 00 00 00]
[S]					 <<< []
[R]				 => [00 01 2B 80  00 00 1D F3]
	id of the camera: 12b80
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 04
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 04 ... (232 bytes more)]
	init bufmask=0x0
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 04
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 04 ... (232 bytes more)]
[C]		ipslr_cmd_10_0a(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 10, a, 4)
[S]					 >>> [F0 24 10 0A  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_cmd_00_05()
[C]			command(fd=3, 0, 5, 0)
[S]					 >>> [F0 24 00 05  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [B8 00 00 00]
[C]			read_result(0x3, size=184)
[S]					 >>> [F0 49 00 00  B8 00 00 00]
[S]					 <<< []
[R]				 => [80 00 1F FF  80 00 03 F1  00 00 00 07  80 00 00 AF
				     00 00 00 03  00 00 00 07  80 00 00 01  80 00 00 01 ... (152 bytes more)]
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 05
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 06 ... (232 bytes more)]
[C]	pslr_camera_name()
./pktriggercord-cli: GX-1L Connected...
[C]	pslr_get_status()
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 05
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 06 ... (232 bytes more)]
[C]	pslr_get_status()
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 05
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 06 ... (232 bytes more)]
[C]		ipslr_read_dspinfo()
[C]			command(fd=3, 1, 1, 0)
[S]					 >>> [F0 24 01 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [04 00 00 00]
[C]		ipslr_read_dspinfo() bytes: 4
[C]			read_result(0x3, size=4)
[S]					 >>> [F0 49 00 00  04 00 00 00]
[S]					 <<< []
[R]				 => [01 00 00 00]
Firmware version: 1.00.00.00
[C]	pslr_get_settings_json()
[C]		ipslr_read_setting(0)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x0})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 00]
[C]			command(fd=3, 20, 9, 4)
[S]					 >>> [F0 24 20 09  04 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
^C

tempUser99, 2018-12-22:
Note: When running both `sudo ./pktriggercord-cli --debug` and `pktriggercord-cli --status_hex --debug` the top LCD on the camera changes from `Pc` to information that is usually there when not connected to USB like is ready to take a picture. In fact, after running either of those commands above the camera can take pictures as if it were not connected to USB. Usually when the LCD on the top of the camera says `Pc` the camera cannot do anything from the actual physical buttons on the camera (like taking pictures, viewing pictures, etc). When terminating the command the camera stays in a state in which it acts like it is not connected to USB unless I power cycle the camera at which time it changes back to `Pc` until I run either of the commands above.

tempUser99, 2018-12-27:
I am going down this path a little further and looking into when we are getting an `[R]	=> ERROR: 0x00` response at one point. I am feeling like that might be the time when the camera appears to seemingly switch over to the second mode where it is able to take a picture.

Do you think that is the right direction to investigate?

asalamon74, 2019-01-01:
@tempUser99 A few tips:

- The program stuck at settings reading. Since this camera model does not have a settings parser, you might comment out the line ( https://github.com/asalamon74/pktriggercord/blob/master/pktriggercord-cli.c#L998 ) because settings info is not that import. Quite likely the program will stuck at a later step, but it's worth trying.

- Yes it's possible that terminating the command leaves the camera in an invalid state, but I only seen things like that in newer models when the camera was not mounted.

- Starting the program does not really disables all the controls of the camera. For some models it is possible to start the program using the `--noshutter` switch. The program connects to the camera but waits for a shutter press and reads the image after that. So it's possible that this camera is also in this state.

- It is possible that the camera is  not in the correct state, maybe pslr_connect ( https://github.com/asalamon74/pktriggercord/blob/master/pslr.c#L426 ) needs to be modified.

tempUser99, 2019-01-04:
We are indeed getting somewhere now. Although nothing is happening on the camera at least the camera does stay in `Pc` mode. Below is the output I get after commenting out the line you suggest (998).

pi@raspberrypi:~/Desktop/pktriggercord_PS $ sudo ./pktriggercord-cli --status_hex --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli --status_hex --debug 
./pktriggercord-cli 0.84.05 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
	Checking drive:  sg0 Samsung 
 Digital Camera  

	Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 00 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 04]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 00 00 00]
[C]			read_result(0x3, size=8)
[S]					 >>> [F0 49 00 00  08 00 00 00]
[S]					 <<< []
[R]				 => [00 01 2B 80  00 00 1D F3]
	id of the camera: 12b80
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
[... 205 lines cut ...]
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_set_mode(0x0)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x0})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 00]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 00 04]
[C]	pslr_shutdown()

asalamon74, 2019-01-05:
@tempUser99 This one actually looks very good. Then program was able to connect to the camera and read the status buffer. Since status parsing is not implemented yet, the fields are not rcorrectly ecognized.

You can improve it in two ways:

1. Modify the GX-1L line, add buffer size info (264 bytes) and create a parser for this camera. It might be very similar to ipslr_status_parse_istds. Most of the fields are zeroes, but you can probably find basic fields like aperture, shutter speed.

2. Try to take a picture. Theoretically it should work even without the status parser.

tempUser99, 2019-01-05:
What is the correct procedure to attempt to take the picture without the buffer?

I will see what I can do to create a parser.

asalamon74, 2019-01-05:
The following command should take a single picture and save it:

```
pktriggercord-cli -o test.pef --debug
```
The output file format will be most likely a pef for your model. Maybe a dng.

tempUser99, 2019-01-18:
@asalamon74 Do you have any updated thoughts based on the last output? I have been making a few modifications to try to test but nothing appears to be helpful.

asalamon74, 2019-01-20:
It seems to me that the shutter command (this command tells the camera to take a picture) is unrecognised, and gives and error:

```
[C]	pslr_shutter()
[C]		ipslr_press_shutter(fullpress = true)
...
	ERROR: 0x82
		shutter result code: 0x82
```

This command is the same for all the camera models, so I don't really think we need to change this part of the code. Maybe the camera initialization code should be different. You may try to change the code, but this is only a trial and error approach.

For some camera models, upgrading the firmware was helpful. According to the logs, your are using firmare 1.00 (or firmware version reading is not working), upgrading it might help.

tempUser99, 2019-01-21:
I managed to get what looks like the only available copy of 1.02 firmware and I upgraded. Now this is all I see when running:

`sudo ./pktriggercord-cli --status_hex --debug`

```Debug messaging is now enabled.
command line:
./pktriggercord-cli --status_hex --debug 
./pktriggercord-cli 0.84.05 
model (null)
device (null)
[C]	plsr_init()
driveNum:27```

I am working on getting passed this point, any thoughts?

asalamon74, 2019-02-01:
That is quite strange. After printing the `driveNum` the program reads info from the `/sys/class/scsi_generics` or `/sys/block` directory, and prints out the `Checking drive...` text which is missing from your output. This should be in the log, because it is printed before we start the real communication. When we check the drives, we might check lots of other devices, some of them are not even cameras.

I'll add a few debug messages soon.

asalamon74, 2019-02-08:
I've added more debug messages to the code. @tempUser99 can you please try it again.

tempUser99, 2019-02-24:
I finally was able to get connected again and ran a test with the new debug messages you added. See attached below.

```
pi@raspberrypi:~/Desktop/pktriggercord_DEBUG $ sudo ./pktriggercord-cli --status_hex --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli --status_hex --debug 
./pktriggercord-cli 0.85.01 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
Getting drive info for sg0
Looking for vendor
vendor: Samsung 

Looking for model
model: Digital Camera  

Looking for device file
	Checking drive:  sg0 Samsung 
 Digital Camera  

	Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 02 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 02 04]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[... 291 lines cut ...]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 02 04]
[C]	pslr_shutdown()
```

tempUser99, 2019-02-24:
Additionally when attempting to run `sudo ./pktriggercord-cli -o test.pef --debug` the following is the output.

```
pi@raspberrypi:~/Desktop/pktriggercord_DEBUG $ sudo ./pktriggercord-cli -o test.pef --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli -o test.pef --debug 
./pktriggercord-cli 0.85.01 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
Getting drive info for sg0
Looking for vendor
vendor: Samsung 

Looking for model
model: Digital Camera  

Looking for device file
	Checking drive:  sg0 Samsung 
 Digital Camera  

	Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 02 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 02 04]
[C]		ipslr_identify()
[C]			command(fd=3, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x3)
[... 397 lines cut ...]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
^C
```

asalamon74, 2019-02-27:
@tempUser99 The strange error of printing out only the driveNum disappeared, right now it's possible to connect to the camera again.

Unfortunately I don't see too much difference in the logs. Status buffer reading works, taking a picture gives the same `ERROR: 0x82` error.

tempUser99, 2019-02-28:
I ran a few more tests, without changing anything to try to better understand what is happening.
Output appears to be the same for ```sudo ./pktriggercord-cli --status_hex --debug``` but for ```
sudo ./pktriggercord-cli -o test.pef --debug``` it appears to be looping differently.

```0x82``` appears to mean that the camera is rejecting the command but at least it is responding to something.

```
pi@raspberrypi:~/Desktop/pktriggercord_DEBUG $ sudo ./pktriggercord-cli -o test.pef --debug
Debug messaging is now enabled.
command line:
./pktriggercord-cli -o test.pef --debug 
./pktriggercord-cli 0.85.01 
model (null)
device (null)
[C]	plsr_init()
driveNum:27
Getting drive info for sg0
Looking for vendor
vendor: Samsung 

Looking for model
model: Digital Camera  

Looking for device file
	Checking drive:  sg0 Samsung 
 Digital Camera  

	Found camera Samsung 
 Digital Camera  

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 01  00 00 00 0E  00 0C 01 20  03 00 00 04]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=3, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=3, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  03 00 00 04]
[... 297 lines cut ...]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  84 00 00 05
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 05 ... (232 bytes more)]
	p->status.bufmask = 0
	No buffer data (0)
get buffer 0 type 5 res 0
[C]	pslr_buffer_open(#0, type=5, res=0)
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 264
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 0E  00 0C 02 20  84 00 00 05
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 05 ... (232 bytes more)]
	p->status.bufmask = 0
	No buffer data (0)
get buffer 0 type 5 res 0
```

andyboeh, 2019-08-17:
Did you ever proceed in getting this to work?
I found my old *istDL2 and I'm trying to get it to connect - I added the model info before I found out that it is similar to the GX-1L and I'm experiencing the very same behavior! Mine is also on Firmware 1.02.

asalamon74, 2019-08-17:
No update on this issue.

*istDL2 is basically the same as GX-1L, so they should behave similarly.

andyboeh, 2019-08-17:
OK - so I did som research: I tried the new Pentax SCSI protocol instead of the old one and - the shutter fires. Now I need to figure out how to transfer an image, this still doesn't work.

For now, it reports the following:

```
[C]	pslr_shutter()
[C]		ipslr_press_shutter(fullpress = true)
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 00  02 00 02 20  85 00 00 01
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 05 ... (232 bytes more)]
		before: mask=0x0
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x2})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 02]
[C]			command(fd=3, 10, 5, 4)
[S]					 >>> [F0 24 10 05  04 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
		shutter result code: 0x0
[C]	pslr_get_status()
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 00  02 00 02 20  85 00 08 01
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 05 ... (232 bytes more)]
get buffer 0 type 0 res 0
[C]	pslr_buffer_open(#0, type=0, res=0)
[C]		ipslr_status_full()
[C]			command(fd=3, 0, 8, 0)
[S]					 >>> [F0 24 00 08  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [08 01 00 00]
	read 264 bytes
	expected_bufsize: 0
[C]			read_result(0x3, size=264)
[S]					 >>> [F0 49 00 00  08 01 00 00]
[S]					 <<< []
[R]				 => [00 00 00 02  00 00 00 00  02 00 02 20  85 00 08 01
				     00 00 00 00  00 00 00 00  00 00 00 01  00 00 00 05 ... (232 bytes more)]
	p->status.bufmask = 0
		Select buffer 0,0,0,0
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x0, 0x0, 0x0, 0x0})
[S]					 >>> [F0 4F 00 00  10 00 00 00]
[S]					 >>> [00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00]
[C]			command(fd=3, 2, 1, 10)
[S]					 >>> [F0 24 02 01  10 00 00 00]
[C]			get_status(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => ERROR: 0x00
[C]		ipslr_buffer_segment_info()
[C]			command(fd=3, 4, 0, 0)
[S]					 >>> [F0 24 04 00  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00]
	Waiting for segment info addr: 0x0 len: 0 B=0
[C]			command(fd=3, 4, 0, 0)
[S]					 >>> [F0 24 04 00  00 00 00 00]
[C]			get_result(0x3)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< []
[R]				 => [10 00 00 00]
[C]			read_result(0x3, size=16)
[S]					 >>> [F0 49 00 00  10 00 00 00]
[S]					 <<< []
[R]				 => [00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00]
```
and this repeats until pslr_disconnect() is called. 

The entry in pslr_model.c now looks like the following:

```
{ 0x12b7e, "*ist DL2",     false,  true,  true,  false, false, false, 0,   3, {6, 4, 2}, 5, 4000, 200, 3200, 200, 3200, PSLR_JPEG_IMAGE_TONE_BRIGHT, false, 11, NULL},
```

andyboeh, 2019-08-17:
Actually, my settings are similar to the K110D. I couldn't find any references as to why the K110D doesn't work, only that it's not working. Do you know more?

asalamon74, 2019-08-26:
@andyboeh I don't remember the exact problem with K110D.

Maybe you could try to change 'bufmask_command' to 'true'.