asalamon74/pktriggercord #37: K-3II not working in bulb mode

mccodesurfer, 2019-01-20:
Is there any chance of fixing this. I'm looking at it but not much a developer but I have a K-3II for testing. There's a number of people in the astrophotography community who really need this feature. (K-3II has built-in astrotracer that can have control in bulb mode which may be confusing things.) thanks.

mccodesurfer, 2019-01-20:
I just cloned the latest code, built and ran it - same thing if I svn checkout the latest trunk.

With the camera in bulb mode (astrotracer off) - the shutter operates for a split second. With astrotracer on, it triggers for whatever the astrotracer exposure setting was.

Your link says all this was resolved with astrotracer on, set time, astrotracer off but I don't see that operation.

asalamon74, 2019-01-20:
Can be paste the output of the command you tried to execute (plese also use the `--debug` switch).

mccodesurfer, 2019-01-20:
Ok, I tried it again now that I understand your link and I see I have to physically turn on the astrotracer function in the camera and then pktriggercord will "enable/disable" it to set the exposure. So this worked.

I guess there is no problem of the camera trying to actually track the stars since its off for the exposure. The K-3II operation is a bit convoluted. For people with tracking mounts, they just need good old fashioned bulb mode. For those using astrotracer with tripod, they need the new control with astrotracer enabled for the exposure - is this settable? If not, I imagine its a rare person who would want to use astrotracer tethered because it requires a lot of hand manipulation, for the calibration for example.

Anyway, thanks so much for this. I have to see now if the GPhoto2 boys have imported your latest! GPhoto2 is used by Kstars/Ekos.

By the way, http://pktriggercord.melda.info/ still says bulb mode does not work on K-3II

asalamon74, 2019-01-20:
The "old" bulb command is not working on newer cameras. Interval bulb timer can be used in K-70 (and in a few other models), but K-3II has no interval bulb timer. The similar astrotracer feature is used in this case. Probably it's not a perfect solution, but this is the best working.

I didn't know that you had to physically turn it on in the camera. The whole idea was to turn it on before exposure, set the timing, press the shutter, turn it off.

The homepage contains that information because in the latest release it's not working. I should create a new release soon.

mccodesurfer, 2019-01-20:
Yeah, the astrotracer function has a two stage on/off control :-s  Anyway your solution seems to be working well. I'll advise the photo2 guys of your upcoming release. Thanks!

asalamon74, 2019-01-27:
I released v0.85.00

Closing this issue as a duplicate of #25

closed 2019-01-27 by asalamon74