asalamon74/pktriggercord #38: make fails - sg.h not found

mccodesurfer, 2019-02-04:
Hi, There seems to be a missing file in the repository. make fails with "scsi/sg.h not found" after cloning the repo and running make on a Mac.

asalamon74, 2019-02-05:
There is no [SCSI Generic](http://www.tldp.org/HOWTO/SCSI-Generic-HOWTO/) driver for Mac, that's why you get this error message.

And this is the reason why Mac is not supported.

asalamon74, 2019-02-05:
Maybe they are using some different kind of protocol. Can you share a link to the USB driver you are referring to.

mccodesurfer, 2019-02-05:
I really don’t know much here. @rlancaste of indilib is going to going to look into it. He’s the expert. He may have something.

I’ve found some objective C code for USB access that I thought I could make a library out of, and provide a dylib. I’ve been able to access my Pentax K-5II with it but it would need some “scsi"_read/write functions to be written around it.

rlancaste, 2019-02-05:
Yes, the INDI GPhoto driver is built on libgphoto2.  This library builds on Linux, Mac, and even Windows.  So we are able to provide INDI support on Macs and Linux for Canon, Nikon, and any other cameras that libgphoto2 supports.  My understanding is that the library talks directly to the camera to instruct it to take exposures and that the issue with Pentax is that it doesn't include the ability to take exposures this way.  Is that correct?

rlancaste, 2019-02-05:
Does PKTrigger Cord completely control via USB?

rlancaste, 2019-02-05:
So I am guessing that PKTriggerCord uses a different type of connection to the camera than libgphoto2?

asalamon74, 2019-02-06:
It seems that modifying the [bsd](https://github.com/asalamon74/pktriggercord/blob/master/pslr_scsi_openbsd.c) port of the program is more promising. With a simple hack I was able to compile it on a Mac. It still has lots of warnings and need testing. I'll investigate it soon.

mccodesurfer, 2019-02-06:
If you think you can gain access to the Mac USB then I'll hold off making a dylib. Let me know as soon as you can. We may incorporate pktriggercord directly into Kstars and bypass gphoto2 for Pentax.

Blueshawk, 2019-02-07:
> Yes, the INDI GPhoto driver is built on libgphoto2. This library builds on Linux, Mac, and even Windows. So we are able to provide INDI support on Macs and Linux for Canon, Nikon, and any other [...]

I don't have a Mac but maybe I can help explain the state of things. The way the indi_pentax driver works is ekos>>indi>>libgphoto>>pktriggercord. The indi driver wraps to libgphoto which is in turn wrapping to PkTriggercord for calls to the camera using "MSC" (usb:scsi) through the usb port to the camera instead of the PTP used by the other brands. 
Cheers, 
Ray

rlancaste, 2019-02-07:
Ok, does libgphoto2 run pktriggercord as a separate program where it executes commands or does it use it as a library with functions that libgphoto2 calls?

asalamon74, 2019-02-07:
They import the source code of pktriggercord, you can find it here: https://github.com/gphoto/libgphoto2/tree/master/camlibs/pentax

They sync the source code sometimes. Occasionally there ask for changes in pktriggercord to make it easiere to import it, but it's quite rate. For instance: #22

Blueshawk, 2019-02-07:
ack..sorry! I forgot to send this. Went looking for the link and got busy. 

> Ok, does libgphoto2 run pktriggercord as a separate program where it executes commands or does it use it as a library with functions that libgphoto2 calls?

Yeah it's library calls, and the rest of the program function is similar to other brands represented. If I remember right some output that would come to screen has been parsed out, but it's not a scripted call like you might do in bash. Gphoto is modular so you can see the Pentax code pretty easily.

rlancaste, 2019-02-07:
Ok I got it, so the pktriggercord code in libgphoto2 just goes into one of the camlibs in the camlibs folder.  You don't have to launch a separate program that gets called.  Thank you for the clarification.

mccodesurfer, 2019-02-08:
Thanks @asalamon74 . I looked at the camlibs/pentax folder and it doesn't include USB/SCSI driver so gphoto2 guys must provide one, which makes me wonder why its not working on the Mac, unless they also use a generic scsi driver. I'll see if I can find it but so far its looking like I'll have to write one.

asalamon74, 2019-02-11:
I created a hack to compile for Mac OS X. The macosx branch contains it: https://github.com/asalamon74/pktriggercord/tree/macosx

Basically I tried to use the BSD port of the program. Since `scsiio.h` was missing I copied the file from BSD. It is possible to compile the program. GTK is a bit tricky so it's better to compile it using `make cli`.

When I connect my camera (K-x) the system recognized it, at least I can get info using `ioreg -p IOUSB -l`. The BSD port was assuming `/dev/rsd*` device which is mising from Mac OS X. The new devices after I connect the camera: `/dev/disk2`, `/dev/disk2s1`, `/dev/rdisk2`, `/dev/rdisk2s1`. I've tried all the devices but only got `Device open while querying:` and `IOCTL failed in query` error messages. The first one can be eliminated by umounting.

I don't know too much about Mac OS X ioctl, so any help would be appreciated.

mccodesurfer, 2019-02-11:
That’s excellent! I have written the ioctl part. I was just trying to test it. Seems to be working but your Mac build will help a lot. Now I can try sending actual camera commands. I copied the scsiio.h also but I think in the end I won’t need it. Question though. My stuff is written in objective C. Can I add it to pk and compile? How to compile with C ?

mccodesurfer, 2019-02-11:
Cause I’m using the Xcode environment. Will gcc recognize objective C?

rlancaste, 2019-02-11:
You could do this in terminal on OS X, it worked fine for me in an experiment:

mkdir ~/Desktop/test
cd ~/Desktop/test
git clone https://github.com/asalamon74/pktriggercord.git
cd pktriggercord
git checkout macosx
make cli

asalamon74, 2019-02-11:
@mccodesurfer Please only use C, not objective C. Gcc might recognize objective C, but we need to compile pktriggercord on different platforms, so the best is to use only ANSI C.

mccodesurfer, 2019-02-11:
May have to provide a library then because the code is not written from scratch. Uses macOS frameworks and libraries. The way I understand it. I just have to put the dylib file in the /usr/local/lib but not sure how to link it in the make file

asalamon74, 2019-02-12:
@mccodesurfer If you provide a library then the language of the library is not that important, only the few calls should be written in C.

Just hack together something with using the -I -L -l switches (-I for the directory containing the headers, -L for the directory containing the lib, -l for the library linking) and I try to clean it later.

mccodesurfer, 2019-02-13:
Ok. Will do. It may take a week or two. I’m not on it as much as I’d like.

mccodesurfer, 2019-02-23:
@asalamon74 I am close with this macOS driver - getting status from camera failing on commands- I haven't been able to relate your USB description "pentax_scsi_protocol.md" to this doc: https://www.beyondlogic.org/usbnutshell/usb6.shtml#SetupPacket

In particular, the pslr_init sets up ioctl for BSD. I have a similar ioctl that uses the setupPacket structure. Your cmd structure is similar (I guess) but I haven't figured out how to form it into the packet structure the macOS uses.

mccodesurfer, 2019-02-23:
Any advice would be great. Thanks

asalamon74, 2019-02-24:
@mccodesurfer 

[pentax_scsi_protocol.md](https://github.com/asalamon74/pktriggercord/blob/master/pentax_scsi_protocol.md) describes a higher level protocol. It describes the byte arrays we send to the camera, the bytes array read from the camera. Probably you don't really need this.

The low level interfaces define `scsi_read` and `scsi_write` methods to send/receive data.

The `pslr_init` tries to find the device and initialize it. Probably you are referring to the `get_drive_info` method called by `pslr_init`. Is that correct? On OpenBSD it uses ioctl, but it's possible that on Mac there are other ways to get the information. For instance the linux version reads the info from the `/sys` filesystem without using ioctl. Unfortunately it seems to me that there is no such filesystem on Mac.

mccodesurfer, 2019-02-24:
Yes, I'm trying to emulate the low level SCSI_read, SCSI_write that call ioctl (in bsd version). In particular, I replace ioctl - this seems to be where the rubber meets the road - with some macOS interfaces. One called "DeviceRequest" use is very similar to ioctl; build a USB byte stream structure with USB byte codes (per the link I gave you) and call DeviceRequest on the structure.

From what you're saying, I guess I need to encapsulate the Pentax protocol into the DeviceRequest protocol. I was thinking they were the same level. They seamed to be the same in many respects, like the F0 and TT codes look like bmRequestType and bRequest. Since F0 code defines vendor specific codes to follow, it looks like this byte stream protocol is USB, not Pentax specific. However, F0 in particular does not exactly match the USB protocol in the link - if F0 where bmRequestType, it would be a reserved code. Just thinking out loud here.

I guess, I have to find out more about using DeviceRequest.

mccodesurfer, 2019-02-24:
I decided to shift the interface with Wireshark on a pc and see that indeed there is 30 bytes or so preamble to the pentax command codes... still working on it!

mccodesurfer, 2019-02-24:
Got Wireshark on Mac now. Looks like I have to make "interface" transactions, which I had working a while ago but its not working right now. Anyway, just letting you know the status.

rlancaste, 2019-02-25:
Good to see you are making progress!  I’m wondering if you have given any thought yet as to whether we want to get the code for controlling this into libgphoto or whether we want to make a new INDI driver for pentax entirely ?  Obviously the library you are writing could work either way, but I’m just trying to plan ahead for when you finish.  Personally I think the best route due to encapsulation and simplicity would be if we can get it into gphoto

mccodesurfer, 2019-02-25:
Thanks Rob! I like the idea of a Pentax driver. Its been almost a year and I don't believe GPhoto2 has yet been updated with some significant fixes in pktethercord. For the future, Pentax has finally come out with an SDK for their cameras but only support releases from the first k-1 and later. But future ports from their SDK could be truly full featured.

tempelmann, 2019-04-01:
Any news on this project? It's not clear to me whether there's so far an actual working Mac version of pktriggercord.

I tried [rlancaste's instructions](https://github.com/asalamon74/pktriggercord/issues/38#issuecomment-462435519), but when I run it with sudo, I only keep getting a "Resource busy" message.

I also wonder if [SIP](https://en.wikipedia.org/wiki/System_Integrity_Protection) has to be turned off for this to work (mine is off).

I'm testing on 10.13.6

rlancaste, 2019-04-01:
Which command gives "resource busy"

mccodesurfer, 2019-04-01:
I'm still working on it...slowly. Ran into a roadblock on the USB access and got some help on Codementor. The guy steered me in a new direction using new high level Apple API for camera access and control, which looks really easy to use but I'm just getting going on it. It seems to be undocumented - very new I guess.

tempelmann, 2019-04-01:
@rlancaste I get this error when I invoke the cmd without arguments.

Still investigating. Am currently getting the project to run in Xcode to make debugging easier.

tempelmann, 2019-04-01:
Here's what I've found out so far:

this:
    fd = open(deviceName, O_RDWR);
fails always (10.13.6) with error "Device busy".
Changing it to 
    fd = open(deviceName, O_RDWR | O_SHLOCK);
makes it work.
However, the next call fails again:
    if (ioctl(fd, SCIOCCOMMAND, &screq) < 0 ||
I have no idea why. I'm stuck.

mccodesurfer, 2019-04-01:
@tempelmann It looks like the API will give access to low level commands that will allow me to tunnel the camera commands through the USB interface. So its experimental.

rlancaste, 2019-04-02:
So just to be clear.  I just test compiled the code.  I cannot actually test the code because I don't have a compatible camera.  Originally the code would not compile, so I was just verifying that it could be compiled.

tempelmann, 2019-04-02:
@rlancaste Oh, bummer then. That means my assessment is correct. Gotta hope @mccodesurfer can get something working, then. Good luck!

mccodesurfer, 2019-05-14:
Well, I'm sorry guys but I've given up on the trying to port to the Mac. It seems every turn ends up a dead end with lack of information. I'm not prepared to put any more time into it. I think the fact that the Pentax is supported on linux is something I can live with. Since Pentax has released new software dev tools, maybe there will be another solution soon.

tempelmann, 2022-02-09:
I had another look after learning how to use Apple's native SCSI APIs:

1. There is really no pass-thru method in macOS via ioctl(), so the BSD access method won't work here.
2. To use the native API (via `<IOKit/scsi/SCSITaskLib.h>`), the device must not be claimed already by a macOS driver. This means that if the Camera is in MSC mode, it'll present itself as a mass storage device, and macOS will take over, and thereby denying access via aforementioned SCSI APIs. Which means that only PTP mode can be used on a Mac, unless one writes a KEXT that prevents macOS from inserting its mass storage driver. But KEXTs won't be possible to make for macOS any longer soon, I'm afraid, so this is a trick business. Still, with older macOS versions (before macOS 12 "Monterey"), a KEXT might be possible, maybe even still thereafter; Apple is quite vague about this. So, this KEXT would not need any code but only claim to be the IOKit handler for the specific Pentax vendor ID. But if that happens, then, I'm afraid, the SCSI APIs won't see the device, and therefore the SCSI API won't be usable either. Which all turns down to the fact that one needs to talk to the Camera via the USB APIs instead, and then talk to it with the SCSI commands. But that's the part I don't understand yet: How does one use the lower level USB protocol to talk SCSI-over-USB? Are there special USB channels involved in that, or do the scsi commands get wrapped somehow?

rlancaste, 2022-02-09:
So, one thing that might help here is that we already disable the PTP Camera program because we have to do that to control Canon cameras and I think other DSLR's as well.  That means that it does not get taken over by MacOS and is left open for gphoto control.  That's already a part of KStars https://github.com/KDE/kstars/blob/ff1ac9cfeb1f11b4fb8725b643ae9e0c8950ab09/kstars/ekos/manager.cpp#L857

I don't know if this will help resolve the issue you mentioned here, but maybe.

tempelmann, 2022-02-10:
@rlancaste Thanks for keeping an eye on this topic. The code you've shown is handling a user-owner process, though, which is easy to deal with (i.e. simply force-quit it) in order to release any hold on the PTP device. With the block storage device this is more difficult because it's macOS (i.e. the kernel) that takes over control of it. And there are no APIs to force it to give it up.

I am currently exploring a different path, though. I have been able to open the endpoints to my K-5 when it's in PTP mode (but not in MSC mode - there I can only open the "usb device" services, but not the "usb interface" services).

Can all Pentax cameras be used with pktriggercard in PTP mode, or do some of them only work with MSC?

rlancaste, 2022-02-10:
No problem, sorry I can't be more help with it, I don't have as much experience working with lower level stuff.  I modified a couple of INDI drivers to make them work better on Macs, but they didn't need the kind of work you are talking about.  But if you get the pktriggercord program working on Mac, I can help by getting it in the 3rdParty build and I can get it building with the rest of the MacOS build for KStars and INDI on craft.  I'm currently working toward the goal of getting it to do nightly builds on the server.

barracuda156, 2025-06-12:
These is some implementation for macOS here: https://github.com/Distrotech/cdrkit/blob/distrortech-cdrkit/libusal/scsi-mac-iokit.c
Not sure if relevant or not (I bumped into the thread looking for something for another issue).

asalamon74, 2025-06-24:
I'm sorry but I cannot develop it without continuous compiling.

barracuda156, 2025-06-24:
> I'm sorry but I cannot develop it without continuous compiling.

GitHub CI support recent macOS versions.

asalamon74, 2025-06-26:
That's a good idea, I'll test it.