asalamon74/pktriggercord #42: Problem setting ISO, shutter speed and aperture on K1

ojalexander, 2019-10-16:
I'm having an issue setting the ISO. The fixed iso value updates and reflects the ISO parameter on the command line but the current iso does not change. It remains set to whatever the camera was last set to using the physical interface.

Setting the aperture and shutter speed works more consistently. I've only had a couple of issues where one or the other did not show up as the current values.

Version:

```
pktriggercord-cli 0.85.01
```

Status:

```
pktriggercord-cli: K-1 Connected...

current iso                     : 6400
current shutter speed           : 5/1
camera max shutter speed        : 1/8000
current aperture                : 3.5
lens max aperture               : 22.0
lens min aperture               : 3.5
set shutter speed               : 5/1
set aperture                    : 3.5
fixed iso                       : 200
auto iso                        : 200-12800
jpeg quality                    : 3
jpeg resolution                 : 36M
jpeg image tone                 : Bright
jpeg saturation                 : 4
jpeg contrast                   : 5
jpeg sharpness                  : 5
jpeg hue                        : 4
zoom                            : 28.00 mm
focus                           : 0
color space                     : sRGB
image format                    : RAW+
raw format                      : PEF
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3 EV steps
custom sensitivity steps        : 1 EV steps
exposure mode                   : 6
scene mode                      : NONE
user mode flag                  : 2
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Select
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket picture counter    : 1
auto bracket ev                 : 0.70
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : HD PENTAX-D FA 28-105mm F3.5-5.6 ED DC WR
battery                         : 7.55V 7.29V 0.00V 0.00V
buffer mask                     : 0000000000000000
```

Command:

```
pktriggercord-cli -t 2 -i 400 -a 5.6
```

Updated status:

```
pktriggercord-cli: K-1 Connected...

current iso                     : 6400
current shutter speed           : 2/1
camera max shutter speed        : 1/8000
current aperture                : 5.6
lens max aperture               : 22.0
lens min aperture               : 3.5
set shutter speed               : 2/1
set aperture                    : 5.6
fixed iso                       : 400
auto iso                        : 200-12800
jpeg quality                    : 3
jpeg resolution                 : 36M
jpeg image tone                 : Bright
jpeg saturation                 : 4
jpeg contrast                   : 5
jpeg sharpness                  : 5
jpeg hue                        : 4
zoom                            : 28.00 mm
focus                           : 0
color space                     : sRGB
image format                    : RAW+
raw format                      : PEF
light meter flags               : 0
ec                              : 0.00
custom ev steps                 : 1/3 EV steps
custom sensitivity steps        : 1 EV steps
exposure mode                   : 6
scene mode                      : NONE
user mode flag                  : 2
ae metering mode                : Multi
af mode                         : AF.S
af point select                 : Select
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket picture counter    : 1
auto bracket ev                 : 0.70
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 0.00
lens                            : HD PENTAX-D FA 28-105mm F3.5-5.6 ED DC WR
battery                         : 7.53V 7.28V 0.00V 0.00V
buffer mask                     : 0000000000000000
```
I really want this to work so that I can streamline someof my processes when I'm shooting at night. Any help would be appreciated. I'd love to help debug this if it's a true issue. Will post a debug output in another message.

ojalexander, 2019-10-16:
[debug_output.txt](https://github.com/asalamon74/pktriggercord/files/3732086/debug_output.txt)

asalamon74, 2019-10-17:
Thanks for the detailed bugreport @ojalexander 

This part of the code is not model-specific, changing the ISO should work the same way for all the models. Unfortunately it also means that I don't really have an idea how to fix the bug.

I do own a Pentax K-x, probably that's one of the best supported model, and changing ISO was not possible in manual mode for that model, I couldn't fix this bug. Which is rather annoying because I usually use M mode.

You might try using ISO intervals (even like 400-400) but to be honest I don't think it will work.