asalamon74/pktriggercord #50: Not working with K10D on Linux

rbnvrw, 2020-05-20:
I have compiled the package manually on ubuntu (make, sudo make install).
When I connect my K10D via USB, I see "PC" on the camera.

When I start pktriggercord from CLI, I get the interface, but it doesn't connect to the camera.
I get the following error message on stderr:
`pslr.c:435:ipslr_status_full(p, &p->status) failed: 4`

Please let me know if you need anything more to debug this issue.

rbnvrw, 2020-05-20:
This is the output with `--debug`:

```
ret: 4
end status_poll
mainwindow_expose
start status_poll
[C]     plsr_init()
driveNum:8
Getting drive info for sg2
Looking for vendor
vendor: PENTAX  

Looking for model
model: DIGITAL_CAMERA  

Looking for device file sg2
        Checking drive:  sg2 PENTAX  
 DIGITAL_CAMERA  

        Found camera PENTAX  
 DIGITAL_CAMERA  

mainwindow_expose
[C]     pslr_connect()
[C]             ipslr_status()
[C]                     command(fd=d, 0, 1, 0)
[S]                                      >>> [F0 24 00 01  00 00 00 00]
[C]                     get_result(0xd)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [10 00 00 00  00 00 01 00]
[R]                              => [10 00 00 00]
[C]                     read_result(0xd, size=16)
[S]                                      >>> [F0 49 00 00  10 00 00 00]
[S]                                      <<< [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00]
[R]                              => [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00]
[C]             ipslr_set_mode(0x1)
[C]                     _ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]                                      >>> [F0 4F 00 00  04 00 00 00]
[S]                                      >>> [00 00 00 01]
[C]                     command(fd=d, 0, 0, 4)
[S]                                      >>> [F0 24 00 00  04 00 00 00]
[C]                     get_status(0xd)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 01]
[R]                              => ERROR: 0x01
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [00 00 00 00  00 00 00 00]
[R]                              => ERROR: 0x00
[C]             ipslr_status()
[C]                     command(fd=d, 0, 1, 0)
[S]                                      >>> [F0 24 00 01  00 00 00 00]
[C]                     get_result(0xd)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [10 00 00 00  00 00 01 00]
[R]                              => [10 00 00 00]
[C]                     read_result(0xd, size=16)
[S]                                      >>> [F0 49 00 00  10 00 00 00]
[S]                                      <<< [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00]
[R]                              => [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00]
[C]             ipslr_identify()
[C]                     command(fd=d, 0, 4, 0)
[S]                                      >>> [F0 24 00 04  00 00 00 00]
[C]                     get_result(0xd)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [08 00 00 00  00 00 01 00]
[R]                              => [08 00 00 00]
[C]                     read_result(0xd, size=8)
[S]                                      >>> [F0 49 00 00  08 00 00 00]
[S]                                      <<< [00 01 2C 1E  00 00 01 A5]
[R]                              => [00 01 2C 1E  00 00 01 A5]
        id of the camera: 12c1e
[C]             ipslr_status_full()
[C]                     command(fd=d, 0, 8, 0)
[S]                                      >>> [F0 24 00 08  00 00 00 00]
[C]                     get_result(0xd)
[S]                                      >>> [F0 26 00 00  00 00 00 00]
[S]                                      <<< [20 01 00 00  00 00 01 00]
[R]                              => [20 01 00 00]
        read 288 bytes
        expected_bufsize: 392
[C]                     read_result(0xd, size=288)
[S]                                      >>> [F0 49 00 00  20 01 00 00]
[S]                                      <<< [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00 
                                              00 00 00 00  00 00 00 0A  00 00 00 00  00 00 00 04]
[R]                              => [00 00 00 02  00 00 00 00  01 00 02 20  03 00 00 00
                                     00 00 00 00  00 00 00 0A  00 00 00 00  00 00 00 04 ... (256 bytes more)]
        Waiting for 392 bytes but got 288
pslr.c:435:ipslr_status_full(p, &p->status) failed: 4
```

rbnvrw, 2020-05-20:
Apologies - after upgrading the firmware from 1.00 to 1.31 it works!

closed 2020-05-20 by rbnvrw

asalamon74, 2020-05-27:
No problem, this firmware requirement was found in #23