asalamon74/pktriggercord #51: K3 Shutter Speed Issue - Firmware 1.40

samwhite59, 2020-05-27:
I'm trying PKtriggercord for the first time with my K3 and running into an issue where the shutter speed can't be set.
Have tried setting different modes on the camera top dial, as well as the GUI and Command Line interface, on Windows 8.1
Camera firmware is the latest, 1.40 - Note this is not a K3-II camera.

GUI always reports the following:
current shutter speed           : 0/0
camera max shutter speed        : 0/0

GUI also reports the following with my camera that may be an issue too?
af point select                 : Unknown value: 6
drive mode                      : Unknown value: 8

Any help appreciated, thanks.

asalamon74, 2020-05-30:
As far as I remember this should work. Things usually work better in the command line version, so I would check that first. Can you send me the output of the following command:

```
pktriggercord-cli --statux_hex --debug
```

samwhite59, 2020-06-02:
There's something wrong here-
C:\Program Files (x86)\pktriggercord-0.85.00-win>pktriggercord-cli --statux_hex --debug

pktriggercord-cli: unknown option -- statux_hex

Running as Administrator, Windows 8.1, pkTriggerCord 0.85.00

samwhite59, 2020-06-02:
Just spotted the typo....

C:\Program Files (x86)\pktriggercord-0.85.00-win>pktriggercord-cli --status_hex -debug
pktriggercord-cli: K-3 Connected...
JSON: Cannot find camera model
0x0000 | 02 00 00 00 00 00 00 00  00 02 00 02 01 00 00 80
0x0010 | 00 00 00 00 03 00 00 00  00 00 00 00 03 00 00 00
0x0020 | 0b 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0030 | 00 01 00 00 00 00 00 00  00 00 00 00 5a 00 00 00
0x0040 | 0a 00 00 00 03 00 00 00  0a 00 00 00 00 00 00 00
0x0050 | 03 00 00 00 0a 00 00 00  00 00 00 00 00 00 00 00
0x0060 | 00 00 00 00 00 00 00 00  c8 00 00 00 64 00 00 00
0x0070 | 80 0c 00 00 00 00 00 00  07 00 00 00 07 00 00 00
0x0080 | 00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x0090 | 01 00 00 00 04 00 00 00  05 00 00 00 05 00 00 00
0x00a0 | 00 00 00 00 01 00 00 00  00 00 00 00 01 00 00 00
0x00b0 | 00 00 00 00 08 00 00 00  00 00 00 00 01 00 00 00
0x00c0 | 01 00 00 00 00 00 00 00  00 40 00 00 01 00 00 00
0x00d0 | 00 00 00 00 00 00 00 00  07 00 00 00 01 00 00 00
0x00e0 | 01 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00
0x00f0 | 01 00 00 00 00 00 00 00  88 13 00 00 04 00 00 00
0x0100 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0110 | 00 00 00 00 5a 00 00 00  0a 00 00 00 03 00 00 00
0x0120 | 0a 00 00 00 00 00 00 00  0a 00 00 00 00 00 00 00
0x0130 | 00 00 00 00 c8 00 00 00  00 00 00 00 00 00 00 00
0x0140 | 0c 00 00 00 38 00 00 00  0a 00 00 00 40 01 00 00
0x0150 | 0a 00 00 00 01 00 00 00  00 00 00 00 50 00 00 00
0x0160 | 0a 00 00 00 00 00 00 00  00 00 00 00 51 02 00 00
0x0170 | be 02 00 00 a4 02 00 00  00 00 00 00 00 00 00 00
0x0180 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0x0190 | 83 00 00 00 00 00 00 00  00 00 00 00 2c 00 00 00
0x01a0 | d0 07 00 00 64 00 00 00  f6 ff ff ff ec ff ff ff
0x01b0 | 00 00 00 00 00 00 00 00  0f 00 00 00 3f 00 00 00
0x01c0 | 00 00 00 00

current iso                     : 200
current shutter speed           : 0/0
camera max shutter speed        : 0/0
current aperture                : 9.0
lens max aperture               : 32.0
lens min aperture               : 5.6
set shutter speed               : 0/0
set aperture                    : 9.0
fixed iso                       : 200
auto iso                        : 100-3200
jpeg quality                    : 4
jpeg resolution                 : 24M
jpeg image tone                 : Bright
jpeg saturation                 : 4
jpeg contrast                   : 5
jpeg sharpness                  : 5
jpeg hue                        : 4
zoom                            : 20.00 mm
focus                           : -10
color space                     : sRGB
image format                    : JPEG
raw format                      : DNG
light meter flags               : 0
ec                              : 0.30
custom ev steps                 : 1/3 EV steps
custom sensitivity steps        : 1 EV steps
exposure mode                   : 6
scene mode                      : NONE
user mode flag                  : 0
ae metering mode                : Center
af mode                         : AF.S
af point select                 : Auto-5
selected af point               : 16384
focused af point                : 0
drive mode                      : Single
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket picture counter    : 0
auto bracket ev                 : 0.30
shake reduction                 : on
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 8.00
lens                            : Sigma or Tamron Lens (3 44)
battery                         : 7.02V 6.76V 0.00V 0.00V
buffer mask                     : 0000000000000011

samwhite59, 2020-06-08:
Did that help at all?

asalamon74, 2020-06-17:
I added a workaround for the camera max shutter speed field, but it will not really help.

One thing I remembered: some of the newer models prefer if you first turn on the camera and connect to the PC only after that. Could you please try that. And please use `--debug` instead of `-debug`.

Oodini, 2021-03-13:
I have the same problem on a K-3(-I) with firmare 1.40.
Connecting the K-3 after switching on resolved the problem, but... then, I cannot change the ISO setting ! But... when I unplug the K-3, the ISO is set to the value I set sooner in PkTriggerCoord.
Also... I cannot change the exposition correction.

56Tbird, 2021-03-31:
I also have a Pentax K-3 with frimware 1.40.
I do not know if my installation went well or I missed bit of information along the way.
It looked like the camera and Lightroom were going to work, but to my dismay the camera went blank and Lightroom would no longer recognize the K-3. I am using the PK Trigger Cord.  I was under the impression that everything was compatible. Any suggestion or installation instructions?