asalamon74/pktriggercord #52: K30, Software shuts down when switched to BULB mode.

lkiersey, 2020-06-13:
[image]

When I switch my K30 to BULB mode the attached message appears and the SW shuts down, If i try to start up in bulb mode, I get the same issue. Its like it cant determine what mode its in. If I start up in U1 mode I can set it to what ever mode I like except bulb, if I set it to bulb the same issue occurs. Other than that functionality is perfect, but I really want to be able to use bulb mode :)  Would really appreciate you taking a look.

Thanks!!
Lenny

lkiersey, 2020-06-15:
made some progress was able to get it to allow bulb mode by setting plsr_model.c line 981 to
`{ 0x12f52, "K-30",        false, true,  **true,**  false, false, false,`

How ever when I take a picture the shutter fires and closes and then the countdown starts so its like it dosent know to hold the shutter open

asalamon74, 2020-07-02:
**Exposure mode reading**

That boolean field was added specially for K-30 because it was the only camera model when the value `false` was required. It's a bit strange that we need to change it to `true`.

Can you please send me the output of the following command (with the original code) when you set the camera to M, Av, B:

```
pktriggercord-cli --statux_hex --debug > debug.log 2>&1
```
**Bulb mode pictures**

Can you please try to take a picture with the cli using `-t 35` to specify 35 seconds. Please try this in B and M mode as well.

lkiersey, 2020-07-04:
Attached are the logs you requested, Manual mode finished at 30 seconds and Bulb mode shutter fired for a split second and failed to remain open.
[35secondManual.log](https://github.com/asalamon74/pktriggercord/files/4872736/35secondManual.log)
[Hex_dump.log](https://github.com/asalamon74/pktriggercord/files/4872737/Hex_dump.log)
[35SecondBulb.log](https://github.com/asalamon74/pktriggercord/files/4872738/35SecondBulb.log)

sh0, 2020-08-10:
@lkiersey can you test pull request #61 and see if it helps you?

lkiersey, 2020-08-12:
> @lkiersey can you test pull request #61 and see if it helps you?

Hi I have tested it and there is a positive and a negative result. The positive is that it recognizes the camera is in Bulb mode and allows me to take a picture both in GUI and CLI mode. The negative is that I still have the same issue where the shutter fails to stay open it fires and closes immediately and downloads the image then after the specified time it completes as if it worked fine.
[image]

so it has always had the issue in the attached image where it cannot get the status. If we can get the shutter to remain open during the exposure then i believe it will be a workable solution.

sh0, 2020-08-12:
@lkiersey Thank you for testing. My changes were mostly about switching to different exposure modes and reading what exposure mode the camera is using. In this sense I guess the patch works to some degree. Can you also tell if the pktriggercord recognizes other exposure modes (Tv, Av, M etc) accurately and if you change modes from pktriggercord then the mode is also correctly changed in camera?

I would suspect that there is also another issue setting camera shutter time. E.g. in bulb mode there needs to be some generic (or maybe special) exposure time setting changed which is currently not done.

lkiersey, 2020-08-12:
Yes I can confirm that the camera correctly switches between all the different modes. And I can select them from the drop down in User mode.

sh0, 2020-08-12:
Thanks. So the mode setting works nicely.

The log lines `pslr.c:981:get_status(p->fd) failed: 129` mean that reading status after sending bulb trigger commands failed. So probably the bulb exposure triggering is done in some other way on K-30.

One quick hack to try is to change camera info flags in `pslr_model.c`. On top of my PR change the following:

    { 0x12f52, "K-30",        false, true, ...

to

    { 0x12f52, "K-30",        false, false, ...

This makes bulb mode use newer commands. However I would suspect that there is a low chance of this patch working as the cameras from this era seem to use older bulb commands.

lkiersey, 2020-08-13:
As you suspected this didnt work as it reports camera cannot work in new bulb mode. Its so close to working :) I can feel it!