asalamon74/pktriggercord #54: Support for 645Z ?

AlvinFrey, 2020-07-28:
Hi everyone !

Is there any testing versions that can work with the Pentax 645Z ?

Here is the output that I get when running pktriggercord cli in debug mode : 

[image]

If that's no the case is there any documentation in order to add my own camera model ?

Best regards,
Alvin

asalamon74, 2020-08-02:
There is no documentation for adding a new camera model.

The first step is to add a single line to the models list like this commit: https://github.com/asalamon74/pktriggercord/commit/4cb385c4c718fd80cb95e1737225de59892b1bea

According to the log you're using Windows. If you're ready to test it, I can create a test version for you.

asalamon74, 2020-08-09:
I created a test version for 645Z:

[pktriggercord-0.85.01-win.zip](https://github.com/asalamon74/pktriggercord/files/5047444/pktriggercord-0.85.01-win.zip)

It should recognize the camera and read the status info. It will not parse the info, first I need same sample. Can you please send the output of the following command:

```
pktriggercord-cli --statux_hex --debug
```

AlvinFrey, 2020-08-10:
Thanks for all ! Sorry for the last answer ! 

I'm going to test this tomorrow and I will come back with those datas.

AlvinFrey, 2020-08-11:
Here is the output of the command : 

[image]