asalamon74/pktriggercord #63: Plans to support KP in the future?

mitsugu, 2020-09-02:
Hi, All.
I can't find KP in the supported cameras listed in README.md. Will you support it in the future?
I'm having trouble because there is no tethering software that works on Ubuntu.

asalamon74, 2020-09-05:
Basic support has been added, but it's not really tested yet. Someone reported that it was possible to take pictures, but bulb mode was not working.

If you want to test it, I suggest to compile the program from the latest source, and start with the command line interface.

mitsugu, 2020-09-05:
Thank you. I will try.

FAb7D, 2022-08-13:
KP don't work. Debug with USB MSC:
pktriggercord-cli --debug
Debug messaging is now enabled.
command line:
pktriggercord-cli --debug 
pktriggercord-cli 0.85.00 
model (null)
device (null)
[C]	plsr_init()
Cannot open /sys/class/scsi_generic
driveNum:38
	Checking drive:  loop1  
	Checking drive:  dm-1  
	Checking drive:  nvme0n1  
	Checking drive:  ram2  
	Checking drive:  ram0  
	Checking drive:  loop8  
	Checking drive:  sdb RICOHIMG
 DSC_KP          

	Cannot get drive info of Pentax camera. Please do not forget to install the program using 'make install'
	Checking drive:  ram9  
	Checking drive:  loop15  
	Checking drive:  loop6  
	Checking drive:  ram14  
	Checking drive:  ram7  
	Checking drive:  loop13  
	Checking drive:  loop4  
	Checking drive:  ram12  
	Checking drive:  ram5  
	Checking drive:  loop11  
	Checking drive:  loop2  
	Checking drive:  dm-2  
	Checking drive:  ram10  
	Checking drive:  ram3  
	Checking drive:  loop0  
	Checking drive:  dm-0  
	Checking drive:  ram1  
	Checking drive:  loop9  
	Checking drive:  loop7  
	Checking drive:  ram15  
	Checking drive:  sda ATA     
 ST1000LM049-2GH1

	Checking drive:  ram8  
	Checking drive:  loop14  
	Checking drive:  loop5  
	Checking drive:  ram13  
	Checking drive:  ram6  
	Checking drive:  loop12  
	Checking drive:  loop3  
	Checking drive:  ram11  
	Checking drive:  ram4  
	Checking drive:  loop10  
	camera not found
diff: 0.001489
sleep 1 sec

asalamon74, 2022-08-13:
Looks like it has no right to read info. Can you try running the program as root?

FAb7D, 2022-08-14:
Great! Work as root!
Anyway the debug is the following:
`sudo pktriggercord-cli --debug
Debug messaging is now enabled.
command line:
pktriggercord-cli --debug 
pktriggercord-cli 0.85.00 
model (null)
device (null)
[C]	plsr_init()
Cannot open /sys/class/scsi_generic
driveNum:38
	Checking drive:  loop1  
	Checking drive:  dm-1  
	Checking drive:  nvme0n1  
	Checking drive:  ram2  
	Checking drive:  ram0  
	Checking drive:  loop8  
	Checking drive:  sdb RICOHIMG
 DSC_KP          

	Found camera RICOHIMG
 DSC_KP          

before connect
[C]	pslr_connect()
[C]		ipslr_status()
[C]			command(fd=0, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x0)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [1C 00 00 00  00 00 01 00]
[R]				 => [1C 00 00 00]
[C]			read_result(0x0, size=28)
[S]					 >>> [F0 49 00 00  1C 00 00 00]
[S]					 <<< [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00 
					      00 00 00 00  00 00 00 00  00 00 00 00]
[R]				 => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 00
				     00 00 00 00  00 00 00 00  00 00 00 00]
[C]		ipslr_set_mode(0x1)
[C]			_ipslr_write_args(cmd_2 = 0x0, {0x1})
[S]					 >>> [F0 4F 00 00  04 00 00 00]
[S]					 >>> [00 00 00 01]
[C]			command(fd=0, 0, 0, 4)
[S]					 >>> [F0 24 00 00  04 00 00 00]
[C]			get_status(0x0)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=0, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x0)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [1C 00 00 00  00 00 01 00]
[R]				 => [1C 00 00 00]
[C]			read_result(0x0, size=28)
[S]					 >>> [F0 49 00 00  1C 00 00 00]
[S]					 <<< [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80 
					      00 00 00 00  00 00 00 00  00 00 00 00]
[R]				 => [02 00 00 00  00 00 00 00  10 02 00 01  00 00 00 80
				     00 00 00 00  00 00 00 00  00 00 00 00]
[C]		ipslr_identify()
[C]			command(fd=0, 0, 4, 0)
[S]					 >>> [F0 24 00 04  00 00 00 00]
[C]			get_result(0x0)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[... 362 lines cut ...]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 01]
[R]				 => ERROR: 0x01
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [00 00 00 00  00 00 00 00]
[R]				 => ERROR: 0x00
[C]		ipslr_status()
[C]			command(fd=0, 0, 1, 0)
[S]					 >>> [F0 24 00 01  00 00 00 00]
[C]			get_result(0x0)
[S]					 >>> [F0 26 00 00  00 00 00 00]
[S]					 <<< [1C 00 00 00  00 00 01 00]
[R]				 => [1C 00 00 00]
[C]			read_result(0x0, size=28)
[S]					 >>> [F0 49 00 00  1C 00 00 00]
[S]					 <<< [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 80 
					      00 00 00 00  00 00 00 00  00 00 00 00]
[R]				 => [01 00 00 00  00 00 00 00  00 01 00 01  00 00 00 80
				     00 00 00 00  00 00 00 00  00 00 00 00]
[C]	pslr_shutdown()
`