asalamon74/pktriggercord #73: K-3 Bulb mode duration not set

jesshurch, 2022-01-13:
Setting camera (K-3) to bulb mode (single press) with a shutter speed of any multiple of seconds results in an exposure of 1/17 of a second.
I get the error:
pslr.c:948:get_status(p->fd) failed: 129
during a wait for the exposure to complete
I first tried this in ekos which uses libpktriggercord (via gphoto2) where setting bulb mode to press-pres meant the exposure would never complete

tested on firmware 1.41 and 1.43 under Opensuse leap 15.3

Blueshawk, 2022-01-14:
I had a very similar problem recently, saw this in issue in the email summary and might be able to help with this. 
  It's not a pktriggercord(wrapped by gphoto, and indilib) error in this case but indiserver. It's been a few weeks now but I'm pretty sure I remember it was because the symlink "indi_pentax_ccd" --> indi_gphoto_ccd was missing in usr/bin. 

sudo ln -s /usr/bin/indi_gphoto_ccd /usr/bin/indi_pentax_ccd
  
 It could possibly be caused by a build script error or permissions during install. 
  If this doesn't do it you might get help at the indilib forum, or 
 set an issue at the indilib git  https://github.com/indilib/indi
  
  Hope this helps!

jesshurch, 2022-01-14:
Thanks Blueshawk,

I have a fairly recent build of libindi (Dec 15) and the symlink is already in place so it doesn't appear to be the case.
I'll try the indilib git as I suspect more people will be using indilib than pktriggercord independently

Blueshawk, 2022-01-14:
I see now I misunderstood something when I responded. I use my k5 in single press mode, and I'm pretty sure at least the pktriggercord/gphoto2 version of the indi driver requires it. 

  The error message and fail to capture originally seemed more in line with my recent error, but I checked the code and it looks like the fail to get status may be because the camera is still busy capturing. I.e In press-press mode indi does a single press, and then releases at timeout, but the camera is still timing waiting for a second press, which then causes the pslr status check failure. 

That just leaves your original timing issue. I'm not sure it matters in the k3 but in the k5 I actually have to set my camera(older k5) to user1 and then set bulb mode in the camera's setup page. I then set the indi driver to B. You might also try enabling "force bulb mode" in the driver. 
Sorry I can't be more help here. Maybe somebody in the indilib forum has a K3 and can help sort it out.

56Tbird, 2022-01-14:
Thank you for your assistance, I have just about given up on the idea of tethering with the K3.I will circle back around an give it a try again.

Blueshawk, 2022-01-14:
Arg... I see now. I wasn't up to speed on the K3 and thought this was solved.  Keep at it. Maybe somebody will shake it loose eventually. FWIW my k50 never did bulb.(gphoto issue 31)

Anyone following can see this issue for more details:
https://github.com/indilib/indi-3rdparty/issues/90