asalamon74/pktriggercord #74: Support for K3iii

ian-morgan99, 2022-06-07:
Has anyone tested yet with a K3iii yet, and if so, is everything working?

asalamon74, 2022-06-12:
Not yet, but I can create a version which handles this camera as the previous K-3 cameras if you can help to test it. What operating system do you use?

ian-morgan99, 2022-06-12:
I'm mostly using Windows with a wired connection, though also starting to use Android Studio to start to learn a bit about Android development , so could try wired in Windows or wifi on Android .

ian-morgan99, 2022-06-12:
PS - Huge thanks :)

asalamon74, 2022-07-20:
First, sorry for the very slow response.

The android version is very limited (and there is no command line version), so I attach a windows version. Basically I treat K-3 III as K-3 II.

First I suggest testing the status reading:

```
pktriggercord-cli --statux_hex --debug
```

If that works, you can try to take a single picture:

```
pktriggercord-cli -o test.dng --file_format=dng --debug
```

[pktriggercord-0.85.01-win.zip](https://github.com/asalamon74/pktriggercord/files/9149689/pktriggercord-0.85.01-win.zip)

ian-morgan99, 2022-07-23:
Brilliant. Many thanks. I'm due to have some time free in the next 2 weeks, so will get testing :)

csantelmann, 2023-02-24:
Is there any hope for a linux version supporting the K-3 III? I could test it on a Debian system.

ian-morgan99, 2023-10-02:
I finally got to test this properly, and alas, I'm getting the error 'Camera not found' when I do the 'status_hex' command.

[C]     pslr_init()
driveNum:24
        Checking drive:  C
        Checking drive:  D
        Checking drive:  E
        Checking drive:  F
        Checking drive:  G
        Checking drive:  H
        Checking drive:  I
        Checking drive:  J
        Checking drive:  K
        Checking drive:  L
        Checking drive:  M
        Checking drive:  N
        Checking drive:  O
        Checking drive:  P
        Checking drive:  Q
        Checking drive:  R
        Checking drive:  S
        Checking drive:  T
        Checking drive:  U
        Checking drive:  V
        Checking drive:  W
        Checking drive:  X
        Checking drive:  Y
        Checking drive:  Z
        camera not found
diff: 19.931952

It seems there is a bit more to do :)

What can I do to help troubleshoot this ?  @asalamon74